raspberrypi/pico-feedback #317: Setting new user expectations

VeloSteve, 2023-03-04:
This is a general comment about the state of Raspberry Pi Pico and the various development environments.

I read some articles about this great little device and rashly assumed that I could start with it as I had with the original Raspberry Pi or Arduino.  Both of those worked right out of the box with pretty easy learning curves.  That is not at all my experience with the Pico W. 

I am wondering whether something could be done to let new users know what they are getting into, ideally before they buy their first device, or at least as soon as they start digging in.  Every new system has to have a startup phase, and that is fine.  Still, it would be valuable know how far along the process is.

Here are some examples of things new users like me would like to know:

- Will I be able to port my existing Arudino projects to Pico?
- What is the state of the IDEs?  Are they ready for use by amateurs and students, or only hardcore hobbyists?  Is one better supported than others?  Is one better for people who want to do very simple things, but another better for more complex work?  Is command line the only thing ready to go?
- Will I be able to read temperature sensors, run PID code, interface with an RTC or LCD Display, and so on?  Are the libraries complete and easy to find, or will I be getting hints from blog posts, downloading untested code, and modifying it?

Just for background here are some of my experiences:

- Started with the **Arduino IDE**, but quickly found that most links online were for MicroPython.  Thought "maybe that is where development is headed".
- Installed **thonny**, which was easy enough.  Even though python isn't my best language, it was easy to run some basic examples and do some coding of my own.  I got as far as running a web server on core 1 while some text continued to print from core 0.
- As soon as I tried to do more, such as reading DS18B20 temperature sensors, I found that there were very few libraries in place in Python.
- Back to the Arduino IDE.  The old OneWire and DallasTemperature libraries don't work, but those from adamboardman/pico-onewire almost did.  I could read one sensor, but not several.  Rolling back a recent "optimization" helped to find the device addresses, but things were still wrong.
- The adamboardman stuff and those using it seemed to prefer a Linux environment.  Cool, I like Linux, just haven't been using it lately.
- Installed Windows Subsystem for Linux with Ubuntu.  This was amazingly quick and easy.
- But using USB to access the Pico within that system was a nightmare.
- Over to my laptop which dual boots to **Ubuntu**.
- Installed a toolchain from instructions on some random site, and their example ran.
- Started to use raspberrypi/pico-examples and got into quite a mess. 

I'm sure I did a lot of things poorly here.  Still, a lot of it would not have happened with some guidance about what to expect.

aallan, 2023-03-05:
I'm really not sure what you're asking for here?

We officially support two development environments for RP2040 and Raspberry Pi Pico; C and MicroPython. Other environments, like the Arduino, CircuitPython, Rust, and many other language choices are available, but are not officially supported.

Overview documentation is available at: https://www.raspberrypi.com/documentation/microcontrollers/

If you want to get started using C, see https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf.

If you want to get started using MicroPython, see https://www.raspberrypi.com/documentation/microcontrollers/micropython.html for a link to the firmware, and https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-python-sdk.pdf for documentation.

closed 2023-03-05 by aallan

VeloSteve, 2023-03-05:
I'm not looking for help here.  I'm suggesting that the public face of the Pico should be more informative and should let users know what they are getting into.  I know that is difficult for a product coming from a mix of business and open source contributors, but it may be worth considering.

I suppose my assumption is that the Pico is ideal for some users and not mature enough for others.  Adoption of Pico will be faster if it can be sold mostly to those who will tell others "this is the coolest new thing", and not to those who will say "the tools are clunky and the libraries aren't in place to do what I could do with an Arduino".  Maybe a little later it can be sold to that latter group, and they will be happy too.

In my case, I'm going to toss the Pico W in a box and try to remember to check it out again in 6 months or so.  Sorry if this was a bit of a rant, but I won't be bothering you for some time.