raspberrypi/pico-feedback #318: Raspberry Pi Pico Python SDK: ADC value different in ADC(channel) vs ADC(Pin(#))

scruss, 2023-03-06:


aallan, 2023-03-06:
Not a documentation issue, so transfered to the `pico-feedback` repo.

scruss, 2023-03-06:
> So just insisting that we use ADC(Pin(#)) doesn't seem like a solution.

Fair enough, but ADC(4) has always seemed to read too high to me, even when I have an accurate temperature sensor sitting on top of the chip. So maybe exposing the internal sensor via a Pin object may not be a bad idea.

There's a PR in for this already, so I hope it goes away soon

lurch, 2023-03-07:
> ADC(4) has always seemed to read too high to me, even when I have an accurate temperature sensor sitting on top of the chip.

See my comment [here](https://github.com/micropython/micropython/issues/10947#issuecomment-1457840879). Do you get the same temperature reading using MicroPython as you do using the C SDK?

aallan, 2023-03-07:
> > So just insisting that we use ADC(Pin(#)) doesn't seem like a solution.
> 
> Fair enough, but ADC(4) has always seemed to read too high to me, even when I have an accurate temperature sensor sitting on top of the chip. So maybe exposing the internal sensor via a Pin object may not be a bad idea.

The on-chip temperature sensor is very sensitive to errors in the reference voltage. If the ADC returns a value of
891 this would correspond to a temperature of 20.1°C. However, if the ADC reference voltage is 1% lower than 3.3V then
the same reading of 891 would correspond to 24.3°C. You would see a change in temperature of over 4°C for a small
1% change in reference voltage. Therefore if you want to improve the accuracy of the internal temperature sensor it
is worth considering adding an external reference voltage.

aallan, 2023-03-14:
Peeking over the wall at micropython/micropython#10947 it looks like this has been, or will shortly be resolved.

closed 2023-03-14 by aallan