raspberrypi/pico-feedback #321: DMA mis-writes when CPU is busy waiting on a PIO fifo status register

dgrinbe2, 2023-03-29:
This is a rather convoluted issue, but it is 100% reproducible. The requirements are a 2GB microSD card, and this: https://www.waveshare.com/pico-restouch-lcd-2.8.htm, with the SD card switched to "SDIO "wiring on the board as documented.

The display and touch are driven using a series of PIO state machines (all in PIO0 unit)

The touch PIO program looks like this (first param is "wait", second is "side set" with 2 bits of data and one bit of enable):

```
	pio0_hw->instr_mem[pc++] = I_MOV(0, 0, MOV_DST_Y, MOV_OP_COPY, MOV_SRC_X);
	pio0_hw->instr_mem[pc++] = I_IN(0, 0, IN_SRC_ZEROES, 9);
	pio0_hw->instr_mem[pc++] = I_WAIT(0, 0, 1, WAIT_FOR_IRQ, 5);
	pio0_hw->instr_mem[pc++] = I_SET(0, 0, SET_DST_PINS, 0);			//nCS low
	lblDoIO = pc;
	pio0_hw->instr_mem[pc++] = I_OUT(1, 5, OUT_DST_PINS, 1);
	pio0_hw->instr_mem[pc++] = I_IN(2, 7, IN_SRC_PINS, 1);
	pio0_hw->instr_mem[pc++] = I_JMP(0, 5, JMP_Y_POSTDEC, lblDoIO);
	pio0_hw->instr_mem[pc++] = I_SET(0, 0, SET_DST_PINS, 1);			//nCS high
	pio0_hw->instr_mem[pc++] = I_IRQ(0, 0, 0, 1, 5);					//sgnal irq and wait for ack
```

Pretty simple: wait for irq5 (send from another SM when we're done driving some display data and moved display nCS high), move touch nCS low, and send some bits spi-like, and recv replies, the bit number configured by "X", then move nCS back up and signal the display SM that we're done.

auto-push and auto-pull are used for tx and rx data, configured as follows, X is also pre-set before SM starts:

```
	//configure sm2 (it uses side enable)
	pio0_hw->sm[2].clkdiv = (3277 << PIO_SM0_CLKDIV_FRAC_LSB);	//less than full speed
	pio0_hw->sm[2].execctrl = (pio0_hw->sm[2].execctrl &~ (PIO_SM0_EXECCTRL_WRAP_TOP_BITS | PIO_SM0_EXECCTRL_WRAP_BOTTOM_BITS | PIO_SM2_EXECCTRL_SIDE_EN_BITS)) | (sm2EndPC << PIO_SM0_EXECCTRL_WRAP_TOP_LSB) | (sm2StartPC << PIO_SM0_EXECCTRL_WRAP_BOTTOM_LSB) | (SIDE_SET_HAS_ENABLE_BIT ? PIO_SM2_EXECCTRL_SIDE_EN_BITS : 0);
	pio0_hw->sm[2].shiftctrl = (pio0_hw->sm[2].shiftctrl &~ (PIO_SM1_SHIFTCTRL_PULL_THRESH_BITS | PIO_SM1_SHIFTCTRL_PUSH_THRESH_BITS | PIO_SM0_SHIFTCTRL_IN_SHIFTDIR_BITS)) | (17 << PIO_SM1_SHIFTCTRL_PULL_THRESH_LSB) | (15 << PIO_SM1_SHIFTCTRL_PUSH_THRESH_LSB) | PIO_SM0_SHIFTCTRL_OUT_SHIFTDIR_BITS | PIO_SM0_SHIFTCTRL_AUTOPULL_BITS | PIO_SM0_SHIFTCTRL_AUTOPUSH_BITS;
	pio0_hw->sm[2].pinctrl = (SIDE_SET_BITS_USED << PIO_SM1_PINCTRL_SIDESET_COUNT_LSB) | (1 << PIO_SM1_PINCTRL_OUT_COUNT_LSB) | (PIN_SPI_MISO << PIO_SM1_PINCTRL_IN_BASE_LSB) | (PIN_LCD_CS << PIO_SM1_PINCTRL_SIDESET_BASE_LSB) | (PIN_SPI_MOSI << PIO_SM1_PINCTRL_OUT_BASE_LSB) | (1 << PIO_SM1_PINCTRL_SET_COUNT_LSB) | (PIN_TOUCH_CS << PIO_SM1_PINCTRL_SET_BASE_LSB);
	
	//give it the value for total number of bits minus one
	pio0_hw->txf[2] = 50;
	pio0_hw->sm[2].instr = I_PULL(0, 0, 0, 0);
	pio0_hw->sm[2].instr = I_OUT(0, 0, OUT_DST_X, 21);	
```

now about the data. First we feed the TX data to the fifo, then we grab the RX data from the FIFO, and then we restart this, forever. This works since the number of data items is 3 for TX and 4 for RX, thus fifos are enough in size and we can use one DMA channel to do all this, in sequence, although another will be needed to re-start all this. I tried BOTH with chaining and with DMA writing the trigger register. Results are the same. DMA is configured as follows:

```
	//touch DMAs have a whole order, and it works thanks to PIOs having 4-word FIFOs
	// * we send the touch constants
	// * we read the touch replies
	// * we re configure ch5 to re-do this again
	struct DmaDescr {
		uint32_t readAddr, writeAddr, xferCt, ctrl;
	};
	static volatile uint32_t mDescrsAddr;
	static volatile struct DmaDescr mDescrs[] = {
		{	//send DATA to touch SM
			.ctrl = (2 << DMA_CH0_CTRL_TRIG_TREQ_SEL_LSB) | (5 << DMA_CH0_CTRL_TRIG_CHAIN_TO_LSB) | (DMA_CH0_CTRL_TRIG_DATA_SIZE_VALUE_SIZE_WORD << DMA_CH0_CTRL_TRIG_DATA_SIZE_LSB) | DMA_CH0_CTRL_TRIG_INCR_READ_BITS | DMA_CH0_CTRL_TRIG_EN_BITS,
			.readAddr = (uintptr_t)mTouchCommands,
			.writeAddr = (uintptr_t)&pio0_hw->txf[2],
			.xferCt = sizeof(mTouchCommands) / sizeof(*mTouchCommands),
		},
		{	//get DATA from touch SM
			.ctrl = (6 << DMA_CH0_CTRL_TRIG_TREQ_SEL_LSB) | (5 << DMA_CH0_CTRL_TRIG_CHAIN_TO_LSB) | (DMA_CH0_CTRL_TRIG_DATA_SIZE_VALUE_SIZE_HALFWORD << DMA_CH0_CTRL_TRIG_DATA_SIZE_LSB) | DMA_CH0_CTRL_TRIG_INCR_WRITE_BITS | DMA_CH0_CTRL_TRIG_EN_BITS,
			.readAddr = (uintptr_t)&pio0_hw->rxf[2],
			.writeAddr = (uintptr_t)mTouchCoords,
			.xferCt = sizeof(mTouchCoords) / sizeof(*mTouchCoords),
		},
		{	//re-configure ch5
			.ctrl = (0x3f << DMA_CH0_CTRL_TRIG_TREQ_SEL_LSB) | (5 << DMA_CH0_CTRL_TRIG_CHAIN_TO_LSB) | (DMA_CH0_CTRL_TRIG_DATA_SIZE_VALUE_SIZE_WORD << DMA_CH0_CTRL_TRIG_DATA_SIZE_LSB) | DMA_CH0_CTRL_TRIG_EN_BITS,
			.readAddr = (uintptr_t)&mDescrsAddr,
			.writeAddr = (uintptr_t)&dma_hw->ch[5].read_addr,
			.xferCt = 1,
		},
	};
	mDescrsAddr = (uintptr_t)mDescrs;
	
	//increments on read and write. write wrapped every 16 bytes so it will keep reconfiguring dma4
[... 152 lines cut ...]

	//todo: consider crc
	return SdHwReadOK;
}
```

No DMA, no touching PIO0

And yet, when this runs, DMA cha 5 is messed up. this happens every time, and is messed up in the same way

I was unable to attach the logic capture that can be opened in "saleae logic" application due to filetype restrictions, so here it is, compressed for size: https://drive.google.com/file/d/15am2WryKCPbimS6Jntkdy8g1AXF3sYHn/view?usp=share_link
I was also unable to attach the test binary, so here that is: https://drive.google.com/file/d/14tfJneBP0ksYPKb-AxqB24ima2hdYR7c/view?usp=share_link

I tried at various CPU clock frequencies between 128 and 192 MHz with no changes

Not doing SD stuff or adding a delay fixes it. Since that it the case, i suspect the looped polling of the PIO status might cause DMA to fail to write to the ch 4 config reg cause the bus is busyy?

I did try to give DMA higher priority than the cpu:

```
//DMA units get wrecked if they do not get bus cycles, to make them high priority
bus_ctrl_hw->priority = (bus_ctrl_hw->priority &~ (BUSCTRL_BUS_PRIORITY_PROC1_BITS | BUSCTRL_BUS_PRIORITY_PROC0_BITS)) | BUSCTRL_BUS_PRIORITY_DMA_R_BITS | BUSCTRL_BUS_PRIORITY_DMA_W_BITS;
```

This made the issue take longer to reproduce (a few more SD commands later) bu totherwise changed nothing

dgrinbe2, 2023-03-29:
I am also happy to be told that I messed up somehow, but I've run out of hair to rip out from my head over this issue, and found nothing.

dgrinbe2, 2023-03-30:
Changing the timing of the state machine 2 on PIO0 helps, though. Slowing it down OR speeding it up.

dgrinbe2, 2023-03-30:
at different speeds i can get it to get a bit further, but it does still die the same way eventually.

kilograham, 2023-03-30:
A guess as to what the problem might be:

```
	{	//re-configure ch5
			.ctrl = (0x3f << DMA_CH0_CTRL_TRIG_TREQ_SEL_LSB) | (5 << DMA_CH0_CTRL_TRIG_CHAIN_TO_LSB) | (DMA_CH0_CTRL_TRIG_DATA_SIZE_VALUE_SIZE_WORD << DMA_CH0_CTRL_TRIG_DATA_SIZE_LSB) | DMA_CH0_CTRL_TRIG_EN_BITS,
			.readAddr = (uintptr_t)&mDescrsAddr,
			.writeAddr = (uintptr_t)&dma_hw->ch[5].read_addr,
			.xferCt = 1,
		},
```

rather than chaining to channel 5, can you instead try using no chaining, and `&dma_hw-ch[5].al3_read_addr_trig` as the writeAddr

If that doesn't work, can you post a whole repro. since you aren't using any of the library helper functions, it isn't clear what setup has and hasn't been done.

dgrinbe2, 2023-03-30:
I will try not chaining to ch5 today. Thank you for looking.
The whole repro is PalmOS, with a lot of modifications :)

Wren6991, 2023-03-30:
First of all, thanks for raising this. We've taken a look at the hardware and have a possible explanation for the behaviour you're seeing:

* The DMA fires a channel's CHAIN_TO (and its completion IRQ) at the point the last write transfer progresses from AHB address phase to data phase
* When an arbiter is contended (in this case the FASTPERI arbiter), completion of address phase at the master port can precede completion of the matching address phase at the slave port, because the request may be buffered whilst waiting for the arbitration grant, due to the operation of the AHB bus handshake
* This means a CHAIN_TO to a channel that you are currently programming can race with the completion of the final write to that channel's registers
* The preferred mechanism in this case is a triggering write, which is guaranteed to trigger the reprogrammed channel at exactly the point that the last programming write completes

Based on this it seems likely that your issue will be fixed by using a triggering write to restart the channel being reprogrammed, like @kilograham said. Please let us know if that doesn't fix things.

Assuming this does fix your issue, we need to do a thorough evaluation offline of what other cases this might affect, and then see what updates need to be made to documentation.

dgrinbe2, 2023-03-30:
Luke and Graham! Thank you so very much! Testing seems to confirm Luke's theory. By removing all cases of "DMA ch N reconfigures DMA ch M, and then chains to M", and replacing them with "DMA ch N reconfigures DMA ch ,such that the last write touches the 'trigger' register, but ch N does not chain to ch M" makes the problem unreproduceable! Thanks for jumping on this so quickly!

kilograham, 2023-03-30:
cough: Graham's theory! ;-)

dgrinbe2, 2023-03-30:
My apologies, Graham.

dgrinbe2, 2023-04-02:
After a few days of torturing this device every which way, I confirm that the fix is complete and correct. Uptime has gone from minutes (dozens of minutes at best) to unlimited (three devices still booted, running, and usable since 3 days ago).

Should I close this issue, or leave it open until doc changes are made?