raspberrypi/pico-feedback #323: Wrong datasheet link for CYW43439 (again)

scvalex, 2023-04-09:
The [link](https://www.infineon.com/dgdl/Infineon-CYW43439-DataSheet-v03_00-EN.pdf?fileId=8ac78c8c8386267f0183c320336c029f) to the CYW43439 datasheet is 404ing again.   This is a repeat of #294.

I spent a bit of time looking, but I can't find any working links.  I also can't find a PDF with that name online or in the wayback machine.

scvalex, 2023-04-10:
I think I found the datasheet on the Mouser website:
- https://eu.mouser.com/ProductDetail/Infineon-Technologies/CYW43439KUBGT?qs=iLbezkQI%252Bsimvupr%252BuwCJA%3D%3D
- https://eu.mouser.com/datasheet/2/196/Infineon_CYW43439_DataSheet_v03_00_EN-3074791.pdf

I have not read it to check if it's complete/accurate/etc.

aallan, 2023-04-10:
It looks like they've moved the datasheet behind a login wall. I'll see if I can find a find a copy elsewhere on their site. Unfortunately, we won't be able to link to the datasheet on a third-party site, in which case we'll probably need to remove the link from our datasheet entirely.

scvalex, 2023-04-10:
That's fair.

Any chance you could mirror it?  If Mouser is allowed to host a copy of it, I assume Raspberry Pi would be to.

aallan, 2023-04-10:
That would be determined by our agreement with Infineon. Mouser is not a manufacturer, they are in a very different position with respect to IP than we are.

closed 2023-04-27 by aallan