raspberrypi/pico-feedback #324: SIE_CTRL register concurrent access

jfedor2, 2023-04-11:
RP2040 Datasheet (ae3b121-clean) has a section ("4.1.2.5.1. Concurrent access") describing how the AVAILABLE bit in the buffer control register needs to be set separately from the rest, after waiting some cycles, to avoid a situation where the USB controller is in the middle of a read and ends up getting a different value to the one that the processor wanted to set.

It appears to me that a similar procedure needs to be followed for the START_TRANS bit in the SIE_CTRL register. Without it I'm getting issues similar to the ones described here: https://github.com/hathach/tinyusb/issues/1830

Applying the following change to TinyUSB makes the issue go away:

```
diff --git a/src/portable/raspberrypi/rp2040/hcd_rp2040.c b/src/portable/raspberrypi/rp2040/hcd_rp2040.c
index 991d0d374..c14099f14 100644
--- a/src/portable/raspberrypi/rp2040/hcd_rp2040.c
+++ b/src/portable/raspberrypi/rp2040/hcd_rp2040.c
@@ -578,6 +578,8 @@ bool hcd_edpt_xfer(uint8_t rhport, uint8_t dev_addr, uint8_t ep_addr, uint8_t *
     uint32_t flags = USB_SIE_CTRL_START_TRANS_BITS | SIE_CTRL_BASE |
                      (ep_dir ? USB_SIE_CTRL_RECEIVE_DATA_BITS : USB_SIE_CTRL_SEND_DATA_BITS) |
                      (need_pre(dev_addr) ? USB_SIE_CTRL_PREAMBLE_EN_BITS : 0);
+    usb_hw->sie_ctrl = flags & ~USB_SIE_CTRL_START_TRANS_BITS;
+    asm volatile ("nop \n nop \n nop \n nop \n nop \n nop");
     usb_hw->sie_ctrl = flags;
   }else
   {
@@ -618,6 +620,8 @@ bool hcd_setup_send(uint8_t rhport, uint8_t dev_addr, uint8_t const setup_packet
   uint32_t const flags = SIE_CTRL_BASE | USB_SIE_CTRL_SEND_SETUP_BITS | USB_SIE_CTRL_START_TRANS_BITS |
                          (need_pre(dev_addr) ? USB_SIE_CTRL_PREAMBLE_EN_BITS : 0);
 
+  usb_hw->sie_ctrl = flags & ~USB_SIE_CTRL_START_TRANS_BITS;
+  asm volatile ("nop \n nop \n nop \n nop \n nop \n nop");
   usb_hw->sie_ctrl = flags;
 
   return true;
```

Assuming this is correct and I'm not just seeing things, this information should probably be added to the datasheet.

aallan, 2023-04-11:
Ping @Wren6991

P33M, 2023-04-11:
Can you provide a minimal selfcontained testcase that compiles against the C SDK and TinyUSB that provokes the issue? Also any scripts that need to be running on the USB host.

jfedor2, 2023-04-12:
Sure.

https://github.com/jfedor2/pico-feedback-324

Put this on a Pico and connect a keyboard or a mouse to it. The LED is on when it's working, when it goes out it means it's broken. Happens randomly, but usually after less than a minute.

P33M, 2023-04-13:
I can get a hang roughly 50% of the time. It depends on the phase of the moon when the board was power cycled. Annoyingly, I can't get it to hang with GDB attached to figure out the register state. My suspicion is that the action the host controller takes on start_trans mis-samples send_setup or (receive_data | send_data) - the start signal is asserted for a single cycle and if all those bits are zero, nothing happens.

P33M, 2023-04-13:
@jfedor2 I think your fix is both necessary and sufficient. Can you submit a pull request against tinyusb latest?

jfedor2, 2023-04-13:
Sure.

Should we add a mention to the datasheet as well?

aallan, 2023-04-25:
The datasheet has been updated and will be published in the next major docs release.