raspberrypi/pico-feedback #328: Some mistakes on raspberry-pi-pico-c-sdk.pdf

MrGreensWorkshop, 2023-04-27:
There are some mistakes at page 41. check with the https://www.raspberrypi.com/documentation/pico-sdk/hardware.html
https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf
wrong                                              correct
pio_set_consecutive_pindirs   ->   pio_sm_set_consecutive_pindirs
sm_config_sideset_pin            ->   sm_config_set_sideset_pins
sm_config_out_shift                ->   sm_config_set_out_shift
pio_sm_enable                         ->    pio_sm_set_enabled

aallan, 2023-04-27:
Not mistakes, but it looks like the API has drifted somewhat since the text was originally written.

MrGreensWorkshop, 2023-04-27:
If there is no mistake, how should I call those? Would you correct me?

lurch, 2023-05-03:
I think Alasdair meant that they weren't mistakes when the documentation was _originally_ written, but the API has been updated and the documentation hasn't yet been updated to match.

aallan, 2023-11-24:
@nathan-contino After thanksgiving can you take a look at this one, it's page 42 in the [published](https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf) version.

aallan, 2023-11-30:
Fixed internally. Will be in the next documentation release.

closed 2023-11-30 by aallan