raspberrypi/pico-feedback #331: bash command line breaks in pdf documenation

topherbuckley, 2023-06-01:
Its trivial for someone familiar with the terminal, but I though it would be worth mentioning as I'm sure a lot of the user-base here is not necessarily terminal-savy. 

In Section 5.1 of the [Getting started with the Raspberry Pi Pico](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf?_gl=1*j5wpj6*_ga*NjYxNTg1MjMuMTY4NTY1MzQ1OQ..*_ga_22FD70LWDS*MTY4NTY1MzQ1OS4xLjEuMTY4NTY1NDkyNy4wLjAuMA..) there is the following command:

```
$ sudo apt install automake autoconf build-essential texinfo libtool libftdi-dev libusb-1.0-0-
dev
```

This should be 

```
$ sudo apt install automake autoconf build-essential texinfo libtool libftdi-dev \
libusb-1.0-0-dev
```

better yet:
```
sudo apt install automake autoconf build-essential texinfo libtool libftdi-dev \
libusb-1.0-0-dev
```
So people can just double click to copy the line rather than select each line with their mouse to ignore the `$` signs on every line. But this might make the line breaks even more confusing. 

Else it is very easy to overlook the fact that it does not install `libusb-1.0-0-dev` and I only realized this when I was trying to configure OpenOCD, and even then the error message was quite easy to overlook. Only when trying to `make -j4` and getting the error message that no targets were available did I back trace this to this. I think this could cause a lot of unnecessary headaches for people. 

I've never contributed to this repo, but I'm not seeing any files associated with the pdf documentation like [Getting started with the Raspberry Pi Pico](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf?_gl=1*j5wpj6*_ga*NjYxNTg1MjMuMTY4NTY1MzQ1OQ..*_ga_22FD70LWDS*MTY4NTY1MzQ1OS4xLjEuMTY4NTY1NDkyNy4wLjAuMA..). Is there another repo for this or are the pdf contents not open-source? Happy to submit a PR if I know where the source for this lies.

lurch, 2023-06-01:
> Is there another repo for this or are the pdf contents not open-source?

Correct, the source-code for the PDFs isn't open source. More information at https://www.raspberrypi.com/licensing/
However I _have_ moved your issue to the `pico-feedback` repo, which is where we track issues with the Pico databook PDFs.

> So people can just double click to copy the line rather than select each line with their mouse to ignore the `$` signs on every line. But this might make the line breaks even more confusing.

Yes, we currently use the `$` to indicate the shell-prompt, which (hopefully) indicates that any command-line **not** starting with a dollar-symbol is actually a continuation of the previous command-line.

topherbuckley, 2023-06-02:
I've never dealt with issue transfers on github. Do I need to close this here or will that close it in the raspberrypi/documentation repo as well?

lurch, 2023-06-02:
> Do I need to...

There's nothing you need to do, it has all been done :slightly_smiling_face:

aallan, 2023-09-05:
This needs doing, but it lower priority.

nathan-contino, 2024-01-18:
Fixed, because we've eliminated those codeblocks entirely in the upcoming release.

closed 2024-01-18 by nathan-contino