raspberrypi/pico-feedback #335: Incorrect path for `texinfo` to build OpenOCD for macOS when installed with `brew` - Appendix A in Getting Started Guide

neerajadsul, 2023-06-27:
Document: Getting Started with Pico, 
Version 2.2 , 14-June-2023

For building OpenOCD on macOS Monterey, we need to install newer version of `texinfo` using `brew`.
After installation the `PATH` is appended with location:

```sh
$ export PATH="/usr/local/opt/texinfo/bin:$PATH"
```
However the **texinfo** `bin` directory is at `/opt/homebrew/Cellar/texinfo/7.0.3/bin`

Therefore correct path export should be as follows.

```sh
export PATH="/opt/homebrew/Cellar/texinfo/7.0.3/bin:$PATH"
```
The brew formula does mention path for macOS in celler. Hope I am not the only one getting the issue.
I can update the doc but not sure it is in this repository as the guide seems to be ISO style quality controlled.

lurch, 2023-06-27:
This might be a difference between Homebrew on an Intel-silicon Mac, and Homebrew on an Apple-silicon Mac? ping @aallan

aallan, 2023-06-27:
This is the difference between Intel and Apple silicon. I thought I'd updated the Mac sections with `NOTE` blocks, but apparently not. It's on the list.

neerajadsul, 2023-06-27:
Ah, yes, it seems so. However, both seems to use the location `Cellar`. Thanks.

aallan, 2023-06-27:
Yup, that's down to your Homebrew version. Earlier versions didn't have that. I'll get it updated.

nathan-contino, 2024-01-18:
Fixed, because we've eliminated the need for to manually build OpenOCD in our upcoming release.

closed 2024-01-18 by nathan-contino