raspberrypi/pico-feedback #336: Content for the cmsis-dap Debug Probe in "Getting Started with Raspberry Pi Pico"

noutram, 2023-06-29:
Regarding: https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf

As the official Debug Probe is now available, it would be helpful to have this formally covered for C/C++ development - especially for use with an IDE such as Visual Studio Code. 

The changes are small but significant (and quite hard to track down!)

**Section 7.3**

Having worked through this document for the first time, it was not clear that pico debug was different to the debug probe. I wasted considerable time trying to unpick the problem. I suspect I will not be the only person to make this incorrect assumption. I wonder if this is worth highlighting?

For the pico debug probe, the key to seamless debug in VS Code seems to be launch.json 
I've raised a PR to have a suitable template added to the examples. However, the change is quite easy (summarised below) and might be worth documenting?

_Changes:_

The file that becomes launch.json would need changing as follows:

* Line 15 would need "interface/raspberrypi-swd.cfg" replaced with "interface/cmsis-dap.cfg"
* To get the correct adapter speed, add the following:

      "serverArgs": [
        "-c", "adapter speed 5000"
      ],

This might save someone a lot of time.

aallan, 2023-06-29:
Yup. That section needs rewriting. It's on the list!

noutram, 2023-06-29:
Ok, let me know if I can help. I spend a fair proportion of my working time writing teaching notes.

(I see you are in Exeter, the town where I was born - working in Plymouth)

noutram, 2023-06-29:
I filed these this morning - might be relevant (depends if they go through)

https://github.com/raspberrypi/pico-examples/pull/401
https://github.com/raspberrypi/pico-project-generator/pull/76

lurch, 2023-06-29:
AFAIK the changes you've identified as being necessary for Debug Probe are _also_ now necessary for the latest version of picoprobe, as that too now uses the standard CMSIS-DAP interface.

noutram, 2023-06-29:
That makes a lot of sense. When I go into the office tomorrow, I'll grab my other pico boards and maybe give that a try if I get a moment. I've only ever used a debug probe with Pico (makes for a tidier desk .. tidy being a relative term of course :)

hannah-m-moore, 2023-09-08:
It is rather satisfying to be able to validate your statement that someone else is having issues with the VSCode while trying to following the introduction to the pico book.  I just posted an issue that I am having with debugging under VSCode...I'm willing to donate anything that I can find out to save someone else spending two weeks fiddling rather than getting on with developing...there is an obvious gap in the documentation somewhere regarding this.

noutram, 2023-09-08:
For now, maybe consider the pico project generator. I submitted a pull request to add support for the pico probe.
https://github.com/raspberrypi/pico-project-generator/pull/76
I hope this helps.