raspberrypi/pico-feedback #337: Raspberry Pi Pico W not recognized as USB Device even in BOOTSEL Mode.

udaykarthikeya7, 2023-07-09:
I have recently bought a Raspberry pi pico W and wanted to learn how to create a web server on it. I first flashed it with the latest firmware from micropython in the bootsel mode and then tried the blink program ( to blink the onboard led every second )in thonny. Then I ran a web server program from this tutorial [https://www.raspberrypi.com/news/how-to-run-a-webserver-on-raspberry-pi-pico-w/] by saving it as main.py

It worked fine for the first few times when I had to make changes to the code in thonny but later when I unplugged the pico and plugged it back again, It is showing as USB device not recognized in my windows 11 pc. After a couple of tries, I tried running the board on bootloader mode by pressing the bootsel button while connecting to the pc using the micro usb data cable. The Same issue persists. USB device is not Recognized. like this,

[image]
[image]

After Searching for a solution for a whole day. I found out that this happens due to the main.py auto running every time the board powers on and a solution could be adding an initial one second sleep in the web server code or renaming main.py to some other name like mywebserver.py but no matter how many times I tried to open the bootloader mode or normally connect the board it neither shows up as a mass storage device nor shows up on thonny so that I connect to the board and change the filename.

But the board isn't broken. It is running the main.py over and over again every time I power the board on. The webserver code is running properly and I'm able to turn on and off the onboard led from the Ip address but I am not able to make any changes or updates as it is not being recognized by my pc. I've tried using other Pcs but The same issue persists. Please Help me solve this issue. I don't have a raspberry pi to program the pico using SWD. All I have is an Arduino uno and a windows 11 pc.

Thank you.

aallan, 2023-07-10:
Putting the board into BOOTSEL mode by **pressing and holding** the button on the board while you plug it into your computer should allow you to reinstall the MicroPython firmware.

Your best source of help for this sort of problem is going to be [our forums](https://forums.raspberrypi.com/viewforum.php?f=143&sid=8f5c81cf09b9662df3ab1656c27e4a57).

closed 2023-07-10 by aallan

lurch, 2023-07-10:
You can also use the `flash_nuke.uf2` file from here https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html#resetting-flash-memory to totally clear the Flash memory (including your old `main.py`, which IIRC doesn't get overwritten when you reflash MicroPython).