raspberrypi/pico-feedback #340: SDK 1.5.1 clock documentation error

matsobdev, 2023-07-27:
There is CLOCKS_CLK_SYS_CTRL_AUXSRC_VALUE_ROSC_CLKSRC instead of CLOCKS_CLK_SYS_CTRL_AUXSRC_VALUE_XOSC_CLKSRC where it suppose to be (it looks like that at least): Examples > Auxiliary Clock Sources > XOSC / clk_sys.

aallan, 2023-08-01:
Sorry, I'm not entirely sure what the question is?

matsobdev, 2023-08-01:
In the 1.5.1 SDK release page on Github, there is HTML documentation https://github.com/raspberrypi/pico-sdk/releases/download/1.5.1/sdk-1.5.1-html-doc.zip and in given location there are two `CLOCKS_CLK_SYS_CTRL_AUXSRC_VALUE_ROSC_CLKSRC` for `XOSC` and `ROSC`. Inside SDK there is `CLOCKS_CLK_SYS_CTRL_AUXSRC_VALUE_XOSC_CLKSRC` for XOSC, that is correct I suppose.

lurch, 2023-08-01:
If this is indeed a "bug", the same thing also occurs at https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#rpipe93dd39da7bd5b73a2e7
(look at the bottom two cells in the right-most column)
[image]

ping @liamfraser (as IIRC he did a bunch of the clocks stuff)

matsobdev, 2023-08-01:
`SDK\src\rp2_common\hardware_clocks\include\hardware\clocks.h` line `57` it is the same - just comment so no huge deal.

lurch, 2023-08-01:
> `SDK\src\rp2_common\hardware_clocks\include\hardware\clocks.h` line `57` it is the same - just comment so no huge deal.

Ah yes, looks like https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_clocks/include/hardware/clocks.h#L57 is the ultimate source of where all these generated-output-files are coming from.

kilograham, 2023-08-02:
cc @liamfraser ; i assume this is just a typo, but haven't looked in detail

closed 2024-11-23 by matsobdev

lurch, 2024-11-25:
Yesh, looks like this was fixed by https://github.com/raspberrypi/pico-sdk/pull/1571 :slightly_smiling_face: