raspberrypi/pico-feedback #342: Update VSCode plugin information

aallan, 2023-08-01:
Update the VSCode plugin information in Section 4.2 of [the MicroPython book](https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-python-sdk.pdf).

See #290. 

[image]

aallan, 2023-09-05:
@nathan-contino This needs poking at to find out where we are with this, since you're going to be installing things in any case this looks like a good one for you. It'd be nice to get a bit more around the VSCode option into the MicroPython book to balance our current coverage of Thonny.

paulober, 2023-10-13:
Any updates on this?

aallan, 2023-10-13:
We've been a bit distracted by [something else](https://www.raspberrypi.com/news/introducing-raspberry-pi-5/) for the last few months. Sorry.

paulober, 2023-10-13:
> We've been a bit distracted by [something else](https://www.raspberrypi.com/news/introducing-raspberry-pi-5/) for the last few months. Sorry. 

I know, sorry, didn't want to annoy, just thought I might have overlooked the change again.

aallan, 2023-10-13:
Not annoyed at all. We've just been… busy. There's a lot of things going on, and this (like anything that wasn't [Bookworm](https://www.raspberrypi.com/news/bookworm-the-new-version-of-raspberry-pi-os/) or [Pi 5](https://www.raspberrypi.com/news/introducing-raspberry-pi-5/) related) has got backlogged. I'm hoping to start in on microcontroller stuff again next month.

aallan, 2024-03-11:
Fixed in the documentation. Will be present in the next release.

closed 2024-03-11 by aallan