raspberrypi/pico-feedback #344: getting-started-with-pico.pdf "9.1. Building on Apple macOS" is not working for Pico-W at least on a Mac M2

Christian-gh-safe, 2023-08-15:
The homebrew toolchain does not include newlib, so nosys.specs and the related libs are missing and some examples fail to compile.

I solved this by installing the current ARM toolchain from https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads, selecting macOS (Apple silicon) hosted cross toolchains / AArch32 bare-metal target (arm-none-eabi) / [arm-gnu-toolchain-12.3.rel1-darwin-arm64-arm-none-eabi.pkg

The pkg install the toolchain in /Applications/ArmGNUToolchain/12.3.rel1/arm-none-eabi

The pico-sdk is able to use it if I create an environment variable:
PICO_TOOLCHAIN_PATH=/Applications/ArmGNUToolchain/12.3.rel1/arm-none-eabi

With this change, I was able to compile all the examples and started my first project.

Thanks for all your products  and the good documentation.

aallan, 2023-09-05:
Last time I checked the instructions worked out of the box as written for both Intel and Arm-based Macs. @nathan-contino Can you take a look? Thanks.

lurch, 2023-09-05:
@nathan-contino As a small bit of background-context, when the Getting-Started-Guide was first written, ARM didn't yet offer toolchain builds for Apple-silicon Macs.

aallan, 2023-09-05:
When it was first being written ARM-based Macs didn't exist.

aallan, 2023-09-05:
> My issue was about PicoW and it's usage of newlib which was not included in the (old) homebrew toolchain for Apple Macos on Arm . The Arm toolchain has newlib and beside the change of toolchain everything is OK.

I understand. However, the last time I did a clean install (a month or two back) the instructions worked as written. We'll will walk through them again and check, and update them if necessary. Thanks!

aallan, 2023-11-20:
Duplicate of #355

closed 2023-11-20 by aallan