raspberrypi/pico-feedback #345: pico w won't reboot

Streakwind, 2023-08-16:
trying to run the blink.uf2 example after I built it in vscode with cmake
if I drag and drop it into the pico w the pico w is ejected and nothing else happens
using windows 11

lurch, 2023-08-16:
Did you build https://github.com/raspberrypi/pico-examples/tree/master/blink or https://github.com/raspberrypi/pico-examples/tree/master/pico_w/wifi/blink ? If you're using a Pico W then you'll need to use the latter. See also https://www.raspberrypi.com/documentation/microcontrollers/c_sdk.html#blink-an-led

closed 2023-08-16 by lurch

Streakwind, 2023-08-17:
@lurch thanks for the help! I believe it works now
however, I still have some problems
I am not able to use cmake to build the program anymore as it does not show up
I also have an error: `command-line error: language modes specified are incompatible`

lurch, 2023-08-17:
@Streakwind The best place for questions like that is https://forums.raspberrypi.com/viewforum.php?f=143