raspberrypi/pico-feedback #346: RP2040 debug / picoprobe documentation

VaporwareII, 2023-08-20:
I just spent several hours fighting to complete the picoprobe instructions in appendix A.  "getting started with Raspberry Pi Pico" Pg 19 still references the wrong branch of OpenOCD, it might have been missed in raspberrypi/documentation#2789 

```
$ cd ~/pico
$ sudo apt install automake autoconf build-essential texinfo libtool libftdi-dev libusb-1.0-0-
dev
$ git clone https://github.com/raspberrypi/openocd.git --branch rp2040 --recursive --depth=1
$ cd openocd
$ ./bootstrap
$ ./configure --enable-ftdi --enable-sysfsgpio --enable-bcm2835gpio
$ make -j4
$ sudo make install
```
--branch rp2040 should either be --branch rp2040-v0.12.0 or default, as that is the new default branch

Additionally, a couple sections on troubleshooting  would really help the first timer understand the failure modes here.

My first stumbling block was I did not know if my USB wire was charge only or if I blew up the pico or what.  being able to break the connections between the Pi, Host A, and Host B into discrete testing sections would make this easier to understand.  The section moves very fast with no examples of test output along the way.
I get it that this duplicates the earlier section of Pi to Pico but there is no way to know as a novice what is particular to Pico vs Pi. I did not have working GPIO on my Pi so I couldn't start with that avenue and work down the tree like the doc is written.

 1. provide a command to prove the pico is connected via USB other than holding down the BOOTSEL and plug/unplugging the board. I don't know enough about the /dev tree to find it in the heirarchy. Maybe picotool?
 2. build picoprobe and install the uf2.  provide a known good version number for different hardware (afaik only pico and pico W?) Provide troubleshooting steps such as checking picotool to see if the normal code loaded instead of the USB mode and you missed holding the BOOTSEL button at the right time. suggest changing to a known good cable.  Point out that you can prove a cable is good by triggering the BOOTSEL on a known good Pico board.
 3. build picotool and provide a command to readback the firmware was installed and it is the correct version to work with openocd in the next section.  sidebar here to mention about Pico W and loop back to build picoprobe for that if they need the wireless chip for some reason.
 4. build openocd, add a reference to copy contrib/60-openocd.rules to /etc/udev/rules.d and add group plugdev to the user. add commands to restart the udevadm. Provide a command to start comms (this is already in the docs appendix A)  Show an example output of Pico Host A unconnected to Pico Host B to prove this worked.
 5. Build Blink Example and load to Host B. (shouldn't be optional you need LED feedback that Host B is powered)
 6. Add SWD wiring to Host B, Run the same openocd command to prove SWD wiring is good.  Show an output example of Host A connected to Host B and remind the user that if there is no comms but the LED is blinking then its a comm failure not power.
 7. Add UART Connection etc

A section about the importance of wiring and how you need a very solid shared GND bus would be beneficial.  I don't know the electrical reasons, but the connection between the SWD ground and the VBUS GND seems to matter.

```
USER@raspberrypi:~/pico/openocd $ src/openocd -f interface/cmsis-dap.cfg -c "adapter speed 5000" -f target/rp2040.cfg -s tcl
Open On-Chip Debugger 0.11.0-g8e3c38f-dirty (2023-08-19-12:33)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 5000 kHz

Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E660B4400720322A
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 5000 kHz
Info : DAP init failed

USER@raspberrypi:~/pico/openocd $ src/openocd -f interface/cmsis-dap.cfg -c "adapter speed 5000" -f target/rp2040.cfg -s tcl
Open On-Chip Debugger 0.11.0-g8e3c38f-dirty (2023-08-19-12:33)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 5000 kHz

Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E660B4400720322A
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
[... 63 lines cut ...]
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
Info : Writing 12288 bytes starting at 0x0
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
** Programming Finished **
invalid command name "verify"
```

lurch, 2023-08-20:
> I just spent several hours fighting to complete the picoprobe instructions in appendix A. "getting started with Raspberry Pi Pico" Pg 19 still references the wrong branch of OpenOCD

This has already been fixed in our internal source, we simply haven't published updated PDFs yet.

> Additionally I think the linebreak in the pdf is being copied on the longer commands lol.

In the documentation-PDFs we use the convention that everything after the `$` symbol should be typed as one long line, i.e.
[image]
should actually by typed in as:
```
openocd -f interface/raspberrypi-swd.cfg -f target/rp2040.cfg -c "program blink/blink.elf verify reset exit"
```

VaporwareII, 2023-08-20:
I thought it was because the $ is at the terminal prompt, but that makes sense.

aallan, 2023-09-05:
Already fixed internally.

closed 2023-09-05 by aallan