raspberrypi/pico-feedback #347: GPIO line 24 seems to be disabled

thangnguyen-69, 2023-08-30:
The original problem is here: 
https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/issues/74
The problem is that without calling the f_open, the GPIO24 line can still output 1. But after calling it, the GPIO 24 line will never be 1 again, even if i try to connect to bluetooth. 
The Pico W datasheet does not seems to talk about this , it only say GPIO24: OP/IP wireless SPI data/IRQ. And nothing more.
How should i proceed.

thangnguyen-69, 2023-08-30:
Is there any other accidental way that may disable the GPIO24?
Where can i find documentation, or code in the SDK that affect this GPIO24?

I read the source for the SD library, but it does not really set the GPIO24 specifically, only setting up SPI interface, and IRQ.

lurch, 2023-08-30:
You might get a quicker response asking for help at https://forums.raspberrypi.com/viewforum.php?f=143

thangnguyen-69, 2023-08-30:
> You might get a quicker response asking for help at https://forums.raspberrypi.com/viewforum.php?f=143

I did actually, the post has been idle in the pico SDK forum thread for 6 days: https://forums.raspberrypi.com/viewtopic.php?t=355486
Maybe i should post it in the General forum instead of the SDK forum.

closed 2023-09-05 by aallan