raspberrypi/pico-feedback #350: macOS install instructions in "Getting started with Raspberry Pi Pico" guide.

trejan, 2023-09-02:
I'm not a Mac user so only reporting an issue from the forum.

"Chapter 9. Building on other platforms" in the latest version of the guide tells you to install arm-none-eabi-gcc from the [ARMmbed/homebrew-formulae](https://github.com/ARMmbed/homebrew-formulae) brew tap. There have been multiple people on the forum complaining about a "arm-none-eabi-gcc: fatal error: cannot read spec file 'nosys.specs': No such file or directory" error when using the latest version of that tap. ARMmbed/homebrew-formulae listed itself as deprecated and to use the [gcc-arm-embedded](https://formulae.brew.sh/cask/gcc-arm-embedded) cask instead. gcc-arm-embedded fixes the nosys.specs error.

The guide says "If you are running on an Apple M1-based Mac you will need to install Rosetta 2 as the Arm compiler is still only compiled for x86 processors and does not have an Arm native version." Is Rosetta 2 still necessary? The [gcc-arm-embedded cask json](https://formulae.brew.sh/api/cask/gcc-arm-embedded.json) has arm64 packages for macOS. If Rosetta 2 is still needed then should change the note to say Apple silicon due to M2 existing now.

lurch, 2023-09-04:
See also #344 and https://github.com/raspberrypi/pico-sdk/issues/1482

aallan, 2023-11-13:
Duplicate of #355

closed 2023-11-13 by aallan