raspberrypi/pico-feedback #351: RP2040 datasheet missing oscillator electrical paramters

ChrisPVille, 2023-09-03:
The RP2040 datasheet is missing many (all?) electrical parameters relating to the XOSC, specifically I'm looking for the transconductance parameter (expressed in A/V or S) of the oscillator amplifier.  Manufacturers either give that value directly (in the case of ST, SiLabs) or provide more detailed info about the crystal maximum safe ESR, bias current, effective drive levels under different conditions, etc. (TI, NXP, Microchip).

This information is critical for selecting a compatible crystal and analytically determining its external current limiting resistor.  I'm interested in calculating the margins on the rp2040 xosc with a different crystal than used on the pico, but don't have any info necessary to select components.

aallan, 2023-09-05:
I'd point you towards the [Hardware Design with RP2040](https://datasheets.raspberrypi.com/rp2040/hardware-design-with-rp2040.pdf) book where we do discuss this topic somewhat. Unfortunately, we can't provide any further information on this topic.

closed 2023-09-05 by aallan

ChrisPVille, 2023-09-05:
I've read though the hardware design pdf you linked, unfortunately there is no useful information about the specifications of the oscillator.  There's a line in section 2.3 saying "You may need to adjust these values if you choose to use a different crystal" however there's no explanation for why a 50 Ohm ESR crystal was chosen nor why a 1K current limiting resistor was chosen.  I could guess it was an estimate derived from the frequency and shunt capacitance of the crystal, but that isn't stated anywhere.

Does this mean these values were only found via experimentation and none of the parameters are documented to calculate them nor determine the oscillator's margin/startup safety factor?