raspberrypi/pico-feedback #355: Compilation on MacOs - M2

HouseSimOn, 2023-10-10:
I've performed steps from https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf to install compiler:
```
$ brew install cmake
$ brew tap ArmMbed/homebrew-formulae
$ brew install arm-none-eabi-gcc
```
After that I tried to compile project and I got following error:
`arm-none-eabi-gcc: fatal error: cannot read spec file 'nosys.specs': No such file or directory`

Then I uninstalled `arm-none-eabi-gcc` using brew and installed `brew install --cask gcc-arm-embedded`. Now compilation works, but I see that compiler identification is `AppleClang` rather than `arm-none-eabi-`. 

Configuring cmake I see correct compilers:
```
% cmake -G "Unix Makefiles" ..
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 12.3.1
-- The CXX compiler identification is GNU 12.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/homebrew/bin/arm-none-eabi-gcc
```

But during compilation I see `AppleClang`:
```
% make 
[  1%] Creating directories for 'ELF2UF2Build'
[  2%] No download step for 'ELF2UF2Build'
[  3%] No update step for 'ELF2UF2Build'
[  4%] No patch step for 'ELF2UF2Build'
[  5%] Performing configure step for 'ELF2UF2Build'
-- The C compiler identification is AppleClang 14.0.3.14030022
-- The CXX compiler identification is AppleClang 14.0.3.14030022
```

Why there is info about `AppleClang` compiler?

DenoG1997, 2023-10-23:
Thanks a lot!!!!!!!!!!!! finally after 12 hours of debugging.......

aallan, 2023-10-23:
@nathan-contino We should possibly look at this as your intro to the PDF toolchain this week.

mjmdavis, 2023-10-24:
FWIW I had the same issue which was resolved by `brew install --cask gcc-arm-embedded`

opsnlops, 2023-10-25:
Same for me. I had to remove the old compiler and `brew autoremove` to let it clean up before I could get the `gcc-arm-embedded` install to go. Once I got the new one installed it was smooth sailing.

axeleckert, 2023-11-02:
Thank you very much! Works after 

1. brew uninstall arm-none-eabi-gcc
2. brew auto remove
3. brew install --cask gcc-arm-embedded

on my Mac M2 Max Sonoma!

aallan, 2023-11-13:
See also raspberrypi/pico-sdk#1543 for OpenOCD related things.

aallan, 2023-11-20:
Doesn't look to be anything specific to M2, or Sonoma. Looks to be a Homebrew imposed thing. General Mac instructions need updating in the Getting Started guide.

aallan, 2023-11-20:
Also see #335

aallan, 2023-12-14:
This one should be closed, right @nathan-contino this got fixed in the recent databooks PR?

nathan-contino, 2023-12-14:
Yes!

closed 2023-12-14 by kilograham

cryptographix, 2024-01-05:
Still not in the latest available pdf

[https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf] is still incorrect

aallan, 2024-01-19:
Documentation release should happen beginning of next week.

taesungh, 2024-02-06:
Thank you for the documentation update @aallan. However, I want to note that tapping `ArmMbed/homebrew-formulae` should not be necessary anymore as [gcc-arm-embedded](https://formulae.brew.sh/cask/gcc-arm-embedded) is a native cask, so I think the documentation in 9.1.1 could be further updated to remove the unnecessary tap
```shell
brew install cmake
brew install --cask gcc-arm-embedded
```

beezbo, 2024-06-28:
Did the instructions get updated? I ran into the same issue trying to follow the getting started guide.

lurch, 2024-07-01:
> Did the instructions get updated? I ran into the same issue trying to follow the getting started guide.

Yes, the current version of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf now says "build-date: 2024-05-02".

If you're still having problems it's probably best to open a new issue.