raspberrypi/pico-feedback #356: RP2040 datasheet: PDF bookmarks only go down 3 levels

esky-software, 2023-10-11:
Bookmarks in the datasheet PDF (and SDK doc) appear to only go down to the 3rd level, even though section numbering goes down further levels.
It would be nice to have all numbered section headings appear in the bookmarks.

aallan, 2023-10-11:
This was a deliberate usability decision, the ToC becomes far too sprawling otherwise.

closed 2023-10-11 by aallan

esky-software, 2023-10-12:
The PDF bookmarks are not the same thing as the table of contents.

lurch, 2023-10-12:
They are in the toolchain (AsciiDoctor-PDF) that we're using.

esky-software, 2023-10-12:
Sounds like the toolchain could be improved.  Ah well.

(Also, the toolchain incorrectly places the Table of Contents bookmark under Colophon bookmark.  C'est la vie I guess.)

esky-software, 2023-10-12:
Also, not true (at least for the latest release of AsciiDoctor-PDF):
https://docs.asciidoctor.org/pdf-converter/latest/pdf-outline/#levels
"You can adjust the depth of section levels that are displayed in the PDF outline independent of the TOC level depth using the outlinelevels document attribute. By default, outlinelevels attribute is set to the same value as toclevels and all the levels are fully expanded."