raspberrypi/pico-feedback #357: ADC FIFO length appears to be eight, not four

drdnar, 2023-10-24:
The RP2040 datasheet clearly specifies, in more than once place, that the ADC has a sample FIFO **four** entries long. However, it *looks* to me to be **eight** entries long. The eight values I read out are unique, the overflow bit doesn't get set until the FIFO level reads eight, and the FSC register hints at an eight-entry queue with its four-bit LEVEL and THRESH fields (yes I know that really just means someone planned for a configurable-size FIFO).

Is this a documentation error? Is there a hardware bug that makes the second half of the FIFO unreliable? Maybe somebody should double-check the VHDL.

I tested this on a B2 stepping, though as far as I know, the only *substantial* differences between steppings are in the ROM.

drdnar, 2023-10-26:
Sample code. I suggest wiring up a potentiometer and turning it back and forth a lot.
```
#include <stdio.h>
#include "pico/stdlib.h"
#include "hardware/adc.h"

int main() {
    stdio_init_all();
    // Change as-needed
    const int ADC_PIN = 28;
    const int ADC_OFFSET = 26;

    printf("\nADC FIFO queue length test\n");

    adc_init();
    adc_gpio_init(ADC_PIN);
    adc_select_input(ADC_PIN - ADC_OFFSET);
    adc_fifo_setup(1, 0, 0, 1, 0); // Enable FIFO, no DMA, no byte shift

    do {
        adc_fifo_drain();
        adc_hw->fcs = (adc_hw->fcs & 15) | ADC_FCS_OVER_BITS | ADC_FCS_UNDER_BITS;
        adc_run(1);
        sleep_ms(1);
        adc_run(0);
        int count = adc_fifo_get_level();
        printf("Count: %i\n{ ", count);
        for (int i = 0; i < count; i++)
            printf("%i, ", adc_fifo_get());
        printf(" }\n");
        sleep_ms(999);
    } while (1);
}
```

Sample output:
```
ADC FIFO queue length test
Count: 8
{ 901, 905, 904, 901, 901, 901, 902, 906,  }
Count: 8
{ 2945, 2945, 2944, 2941, 2945, 2942, 2945, 2947,  }
```

aallan, 2023-10-26:
Ping @dp111, a you spent a lot of time looking at the ADC.

Wren6991, 2023-10-31:
You're correct, the FIFO is 8 samples deep, not 4. Looks like the datasheet documented the default value rather than the as-taped-out value by mistake. This is identical across B0/B1/B2, and the only mistake is the documentation as far as I can see.

aallan, 2023-10-31:
Thanks Luke, I'll get that fixed up.

drdnar, 2023-11-01:
Wow, I had a hard time believing that I was the first person to notice that in two years! But the more I thought about it, it also seemed unlikely to be anything else. This is neat, because it means you can take twice as many samples without having to set up a DMA, which is more convenient if you only wanted eight samples in the first place.

aallan, 2024-01-18:
Fixed in next docs release.

closed 2024-01-18 by aallan

lurch, 2024-02-02:
I just noticed that https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_adc/include/hardware/adc.h also has several places that talk about the ADC FIFO being 4 entries deep, so I guess that needs fixing up too?