raspberrypi/pico-feedback #362: Please

esky-software, 2023-11-16:
See https://github.com/raspberrypi/pico-feedback/issues/356
"Bookmarks in the datasheet PDF (and SDK doc) appear to only go down to the 3rd level, even though section numbering goes down further levels. It would be nice to have all numbered section headings appear in the bookmarks."

As the comment history on that issue shows, the reasons given for closing it are misguided.  The ToC is not the same as the PDF outline, and AsciiDoctor-PDF indeed supports specifying their depths independently:

https://docs.asciidoctor.org/pdf-converter/latest/pdf-outline/#levels
"You can adjust the depth of section levels that are displayed in the PDF outline independent of the TOC level depth using the outlinelevels document attribute. By default, outlinelevels attribute is set to the same value as toclevels and all the levels are fully expanded."

Please reconsider this issue.

nathan-contino, 2024-11-11:
I think you're right: the outline should go deeper than our ToC. Especially in such a large document.

Can't get to this today, but hopefully later this week. Fortunately I already put in the work in January, I just need to make sure it's all compatible with more recent infrastructure improvements.

lurch, 2025-01-04:
@magy00 Please be aware of the "No Derivatives" part of the RP2040 datasheet licence, which means you're not allowed to distribute modified versions.
[image]