raspberrypi/pico-feedback #364: Blink example in Getting Started book

aallan, 2023-11-24:
Chapter 3 of "[Getting Started](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf)" is written for Pico only. Needs to be updated for Pico W.

See this [forum thread](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf) and the board definition files for [Pico](https://github.com/raspberrypi/pico-sdk/blob/6a7db34ff63345a7badec79ebea3aaef1712f374/src/boards/include/boards/pico.h#L33) and [Pico W](https://github.com/raspberrypi/pico-sdk/blob/6a7db34ff63345a7badec79ebea3aaef1712f374/src/boards/include/boards/pico_w.h#L31).

See also the Blink example for [Pico](https://github.com/raspberrypi/pico-examples/tree/master/blink), and for [Pico W](https://github.com/raspberrypi/pico-examples/tree/master/pico_w/wifi/blink).

aallan, 2023-11-24:
@nathan-contino I thought this had been done already, but section 3.1 needs updating to cover both Pico and Pico W on `develop`. Another one for you to look at next week after Thanksgiving.

nathan-contino, 2023-11-27:
Related: https://github.com/raspberrypi/databooks/issues/251. Closing that issue in favor of this one to keep our backlog "clean".

closed 2024-01-18 by nathan-contino