raspberrypi/pico-feedback #365: Make -DBOARD=boardname more prominent in Getting Started and elsewhere

aallan, 2023-11-24:
There is a NOTE block on page 9 that says

> The SDK will build the binary targetting the Raspberry Pi Pico by default. If you are building a binary for a different
board you need to pass -DPICO_BOARD=board where board is the name of the board. For instance if you’re building a
binary for Pico W you should pass, -DPICO_BOARD=pico_w along with information about your wireless network
-DWIFI_SSID="Your Network" -DWIFI_PASSWORD="Your Password".

This needs to be made a lot more prominent up front in the build narrative (now there are more boards than just Pico).

Then reminder NOTE like this need to be scattered through the rest of the text in this book and in [Connecting to the Internet](https://datasheets.raspberrypi.com/picow/connecting-to-the-internet-with-pico-w.pdf) where compiles are done to remind folks about this.

aallan, 2023-11-24:
@nathan-contino Another one for after Thanksgiving.

nathan-contino, 2024-01-18:
Done!

closed 2024-01-18 by nathan-contino