raspberrypi/pico-feedback #370: OpenOCD install error on Ubuntu 22.04

merinkumar, 2024-01-06:
Hi Team, 
     Am trying to install openocd on my ubuntu desktop, following this doc [https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf?_gl=1*56flef*_ga*MTUwNjc3OTc3MC4xNjk4NDIzNTMx*_ga_22FD70LWDS*MTcwNDUzMzMyNy4xMS4xLjE3MDQ1MzQyNTEuMC4wLjA.](url) . Page 61. It mentions the follwing steps.
`Linux
$ cd ~/pico
$ sudo apt install automake autoconf build-essential texinfo libtool libftdi-dev libusb-1.0-0-
dev
$ git clone https://github.com/raspberrypi/openocd.git --branch rp2040 --depth=1
$ cd openocd
$ ./bootstrap
$ ./configure
$ make -j4`

Am able to complete till ./configure . But when i execute make -j4, i get error.. 
`rc/flash/nor/numicro.c -o src/flash/nor/numicro.o
src/flash/nor/numicro.c: In function ‘numicro_fmc_cmd’:
src/flash/nor/numicro.c:1244:17: error: this ‘if’ clause does not guard... [-Werror=misleading-indentation]
 1244 |                 if (retval != ERROR_OK)
      |                 ^~
In file included from ./src/flash/common.h:21,
                 from src/flash/nor/core.h:25,
                 from src/flash/nor/imp.h:24,
                 from src/flash/nor/numicro.c:32:
./src/helper/log.h:119:9: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  119 |         do { \
      |         ^~
src/flash/nor/numicro.c:1246:25: note: in expansion of macro ‘LOG_DEBUG’
 1246 |                         LOG_DEBUG("status: 0x%" PRIx32 "", status);
      |                         ^~~~~~~~~
src/flash/nor/numicro.c: In function ‘numicro_erase’:
src/flash/nor/numicro.c:1513:25: error: this ‘if’ clause does not guard... [-Werror=misleading-indentation]
 1513 |                         if (retval != ERROR_OK)
      |                         ^~
In file included from ./src/flash/common.h:21,
                 from src/flash/nor/core.h:25,
                 from src/flash/nor/imp.h:24,
                 from src/flash/nor/numicro.c:32:
./src/helper/log.h:119:9: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  119 |         do { \
      |         ^~
src/flash/nor/numicro.c:1515:33: note: in expansion of macro ‘LOG_DEBUG’
 1515 |                                 LOG_DEBUG("status: 0x%" PRIx32 "", status);
      |                                 ^~~~~~~~~
src/flash/nor/numicro.c: In function ‘numicro_write’:
src/flash/nor/numicro.c:1602:33: error: this ‘if’ clause does not guard... [-Werror=misleading-indentation]
 1602 |                                 if (retval != ERROR_OK)
      |                                 ^~
In file included from ./src/flash/common.h:21,
                 from src/flash/nor/core.h:25,
                 from src/flash/nor/imp.h:24,
                 from src/flash/nor/numicro.c:32:
./src/helper/log.h:119:9: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  119 |         do { \
      |         ^~
src/flash/nor/numicro.c:1604:41: note: in expansion of macro ‘LOG_DEBUG’
 1604 |                                         LOG_DEBUG("status: 0x%" PRIx32 "", status);
      |                                         ^~~~~~~~~
depbase=`echo src/flash/nor/spi.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
/bin/bash ./libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I.   -I./src -I./src -I./src/helper -DPKGDATADIR=\"/usr/local/share/openocd\" -DBINDIR=\"/usr/local/bin\" -I./jimtcl -I./jimtcl  -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Wpointer-arith -Werror -g -O2 -MT src/flash/nor/spi.lo -MD -MP -MF $depbase.Tpo -c -o src/flash/nor/spi.lo src/flash/nor/spi.c &&\
mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I./src -I./src -I./src/helper -DPKGDATADIR=\"/usr/local/share/openocd\" -DBINDIR=\"/usr/local/bin\" -I./jimtcl -I./jimtcl -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Wpointer-arith -Werror -g -O2 -MT src/flash/nor/spi.lo -MD -MP -MF src/flash/nor/.deps/spi.Tpo -c src/flash/nor/spi.c -o src/flash/nor/spi.o
cc1: all warnings being treated as errors
depbase=`echo src/flash/nor/stmsmi.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
/bin/bash ./libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I.   -I./src -I./src -I./src/helper -DPKGDATADIR=\"/usr/local/share/openocd\" -DBINDIR=\"/usr/local/bin\" -I./jimtcl -I./jimtcl  -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Wpointer-arith -Werror -g -O2 -MT src/flash/nor/stmsmi.lo -MD -MP -MF $depbase.Tpo -c -o src/flash/nor/stmsmi.lo src/flash/nor/stmsmi.c &&\
mv -f $depbase.Tpo $depbase.Plo
make[2]: *** [Makefile:3629: src/flash/nor/numicro.lo] Error 1
make[2]: *** Waiting for unfinished jobs....
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I./src -I./src -I./src/helper -DPKGDATADIR=\"/usr/local/share/openocd\" -DBINDIR=\"/usr/local/bin\" -I./jimtcl -I./jimtcl -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Wpointer-arith -Werror -g -O2 -MT src/flash/nor/stmsmi.lo -MD -MP -MF src/flash/nor/.deps/stmsmi.Tpo -c src/flash/nor/stmsmi.c -o src/flash/nor/stmsmi.o
make[2]: Leaving directory '/home/merin/pico/openocd'
make[1]: *** [Makefile:5318: all-recursive] Error 1
make[1]: Leaving directory '/home/merin/pico/openocd'
make: *** [Makefile:2175: all] Error 2
merin@merin-UBUNTU-7C84:~/pico/openocd$ `

Note: I have instakll pkg-config also as suggested in other posts. And i have the pico sdk setup already and working with example programs. Even in VSCode Platformio setup.

Can you pelase help ?

kilograham, 2024-01-06:
try adding `--disable-werror` as a parameter to `./configure`

merinkumar, 2024-01-06:
> try adding `--disable-werror` as a parameter to `./configure`

thanks @kilograham , after adding that flag to ./configure, in make i dint get any errors. Is it something related to ubuntu updates ? becuase i dint see this flag as an instruction, in any other website...

closed 2024-01-10 by merinkumar