raspberrypi/pico-feedback #372: Add some documentation on pico-sdk memory usage

peterharperuk, 2024-01-29:
e.g. stack usage and how to modify linker scripts.

aallan, 2024-01-29:
Should be heading into the Pico C SDK book? Or is fundamental enough it should end up in the Getting Started book?

peterharperuk, 2024-01-29:
It's probably worth adding a quick description of heap and stack usage goes by default. I think a discussion of linker scripts is very much an advanced topic.

h1771t, 2025-04-23:
Using pico-sdk, could not find any docs on default heap sizes etc. I take it, it is not simple to change a heap size? (not a config change somewhere?)
Use case: using pico_w to make https request, trying to store the response json (20kB or so) to later process it. I'm trying to use realloc as the response comes in. It runs out of memory and fails.
Printing out mallinfo() just before it fails, shows me:
>used/uordblks 22988
>unused/fordblks 5120
>arena 28108

Is 28K some sort of compile time limit or?

lurch, 2025-04-23:
@h1771t You might get a quicker response to your questions at https://forums.raspberrypi.com/viewforum.php?f=143

lurch, 2025-04-24:
...but I've also just noticed that Chapter 5 of https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf mentions a `PICO_HEAP_SIZE` setting on page 534.