raspberrypi/pico-feedback #375: Synchronizing two state machines with one Clock Signal in Raspberry Pi Pico PIO

tybasuleman, 2024-02-01:
Can anyone help me out. I am trying to sync my 2 state machines with a clock signal so when clock is ON, State Machine one sends a signal and when it´s OFF, the other one sends a signal. Signal here is PDM bits. The problem I think I understand is that every instruction is carried out for one complete clock cycle, not half of it as per my requirement and I cannot say out pins, 0.5 or something per say. It has to be between 1 and 32.
So I am unable to turn the state machine off mid way. We cannot use two wait instructions in one pio file as well otherwise that would have done the job. If I am approaching the problem wrong then please correct me.
Thanks

lurch, 2024-02-01:
> The problem I think I understand is that every instruction is carried out for one complete clock cycle, not half of it as per my requirement

Could you run your PIO state machines at twice the frequency of your clock signal?

tybasuleman, 2024-02-01:
That might be a solution, I will try that now but won´t that be sort of hard coding the logic instead of genuinely detecting some condition and acting upon it?

tybasuleman, 2024-02-01:
And I have now tried it as well. And weirdly, data line is not taking the new frequency. It is running on the old one still. Also, This might not be as effective for me since I want the state machines to be turned off when clock is off, so when consecutive 1s will happen, the data line won´t turn off at clock OFF instead it will keep going

lurch, 2024-02-01:
I'm beginning to think that this is a question better suited to https://forums.raspberrypi.com/ than to GitHub.

tybasuleman, 2024-02-01:
I have posted it there too. Thank You :)

lurch, 2024-02-08:
@tybasuleman Is this a commercial project, or merely a hobbyist project?

tybasuleman, 2024-02-08:
It´s related to a course

tybasuleman, 2024-02-09:
Nevermind, It can be solved in the main code instead of handling it in PIO programs.

closed 2024-02-09 by tybasuleman