raspberrypi/pico-feedback #379: [QUESTION] For datasheet documentation typos, how to submit a correction?

shreeve, 2024-02-21:
For datasheet documentation typos, is the best way to file an issue for them to take a screenshot of the documentation and then type in the fix? Or, does the documentation exist somewhere as a markdown (or other) file that a patch or pull request can be made?

shreeve, 2024-02-21:
@lurch - Adding you since you seem to be pretty heavily involved :-)

lurch, 2024-02-21:
> @lurch - Adding you since you seem to be pretty heavily involved :-)

I _used_ to be, but I'm no longer directly involved in the documentation.

The datasheet (PDF) documentation is hosted in an internal repo (it isn't open-source), so your best course of action is to simply list any typos you find in an issue in the pico-feedback repo, and the documentation team will then fix them in due course.

shreeve, 2024-02-21:
@lurch - Thank you! I'll close this issue out.

As a parting question -- Are you aware of *ANY* other documentation for USB Host other than the datasheet and the TinyUSB source? I am working on a TinyUSB-free USB Host library called https://github.com/shreeve/pico-usb and could really use some additional help. It's very close but I could use a little extra help if you are aware of any other documentation for USB Host mode on the rp2040. Thx.

lurch, 2024-02-22:
There's some TinyUSB-free code at https://github.com/raspberrypi/pico-examples?tab=readme-ov-file#low-level-example and https://github.com/raspberrypi/pico-extras and https://github.com/raspberrypi/pico-playground?tab=readme-ov-file#full-applications but I dunno if any of those are what you're looking for?
But I'm (personally) not aware of any other publicly-available documentation around USB Host mode. But as mentioned earlier, I'm not as involved in the documentation as I used to be.

shreeve, 2024-02-22:
Thanks @lurch - The USB Host code in those links still use TinyUSB. I'm chugging along on this alternative right now, I'm *almost* there... once I can fully enumerate multiple devices, I'll remove all the debugging code and will tighten it up so we can have a clean, simple, USB Host example without using any TinyUSB. I'd like to be able to enumerate a variety of devices and be able to show something like `usbdiagnose` shows. Once all that is done, then adding the class drivers should become a much more traditional effort. This initial hardware specific stuff is quite complicated with the materials available now. Perhaps, if I get it running soon, I can put together something helpful. Thanks again! And, sorry to use this closed issue as a conversation thread. Yikes! :-)

lurch, 2024-02-22:
If you have specific questions, you could ask them in a (new) issue here, and if you're _very_ lucky one of our engineers might have time to answer them; but I can't promise anything as we're all always very busy.

shreeve, 2024-02-22:
Sounds great! Thanks for the tip!

shreeve, 2024-02-22:
I found this link: https://github.com/xcd0/rp2040_datasheet/blob/master/src/SUMMARY.md

Is there a way to properly get access to the RP2040 datasheet if we do have some corrections/typos/fixes to make? @aallan - Is that link legit?

aallan, 2024-02-22:
> I found this link: https://github.com/xcd0/rp2040_datasheet/blob/master/src/SUMMARY.md
> 
> Is there a way to properly get access to the RP2040 datasheet if we do have some corrections/typos/fixes to make? @aallan - Is that link legit?

No, it's not. I have no idea where that's from. It is not the source of the datasheet.

We don't make the source, or toolchain, for our PDF-based documentation public. Sorry.

aallan, 2024-02-22:
Our PDF documentation is distributed under CC BY-ND, and that repo is breaking the license. Thanks for pointing it out. I'll follow that up with Github.

closed 2024-02-22 by shreeve