raspberrypi/pico-feedback #380: [BUG?] USB Host raises TRANS_COMPLETE when it shouldn't

shreeve, 2024-02-22:
I am working on a TinyUSB-free USB Host library for the rp2040, called https://github.com/shreeve/pico-usb and am seeing an issue that I can't seem to resolve. I really like the TinyUSB library, but it is very large and pretty confusing. I'd like to create a much simpler, canonical pico-sdk based library.

In the following screenshot, the following registers are defined as:

```
INTR = raw USB IRQ interrupts
INTS = USB IRQ interrupts that we've requested
SSR = sie_status register
SCR = sie_ctrl register
DAR = addr_endp register (device and endpoint address)
ECR = EPX control register
BCR = EPX buffer control register
```

[image]

The issue that I'm having is that in the setup packet being transferred at the top of the image, `bcr` is set to `0x2462` which means that: the next packet is a DATA1 (the leading 2), that buf0 is AVAILABLE (the 4), and I'm requesting `0x62` (98 decimal) bytes. This should raise the `BUFF_STATUS` interrupt when the initial packet completes, but it should **not** raise the `TRANS_COMPLETE` interrupt, since `bcr` did not include `USB_BUF_CTRL_LAST`.

After the initial `0x40` (64 decimal) bytes are received, when I process this initial `BUFF_STATUS`, I set the `bcr` (shown as `~bcr~`) to `0x6422` to indicate that the following packet **will be** the last one (since we only need to transfer the remaining `0x22` (34 decimal) bytes. The leading `6` includes the `USB_BUF_CTRL_LAST` bit (itself being a `4`). I would expect the resulting packet based on the `bcr` of `0x6422` to raise the `TRANS_COMPLETE` interrupt and let me finish this request. However, the **first** packet raises `TRANS_COMPLETE` and I can't figure out why.

The datasheet shows that there are supposed to be only FOUR conditions that raise `TRANS_COMPLETE`:

[image]

I don't think these four conditions apply to the first screenshot above.

My code is found at: https://github.com/shreeve/pico-usb/blob/main/host/host.c

If there's a way to not have `TRANS_COMPLETE` not be raised when it shouldn't, I would greatly appreciate the help. I hope it isn't some lower level issue like one of the few USB errata. My guess is that I'm just doing something wrong, but given that I have found absolutely zero other USB Host libraries for the rp2040 outside of TinyUSB, it's hard to know.

I'm not sure if this is even the correct place to post this question, but @lurch mentioned it might be worth a shot. If @kilograham or @aallan have any input that would be great also! Ideally, if this can be fixed, the code could be helpful to have a more flexible and lighter USB Host library to offer on the rp2040.

Thanks!

aallan, 2024-02-22:
This sounds like a @kilograham question, or potentially @liamfraser as he's been deeply involved USB things, if either has the time right now to look into it.

shreeve, 2024-02-22:
Ah! @liamfraser, that's probably it! If so, then the documentation should be updated to reflect when `TRANS_COMPLETE` will be raised. Is this list correct?

1. A SETUP packet is sent without a following IN or OUT transaction
2. An IN or OUT packet is sent and the `LAST_BUFF` bit is set in the buffer control register
3. An IN or OUT packet is sent with zero length or is a short packet (less than the max packet size)

@aallan - I will see if the short packet was the issue.

shreeve, 2024-02-22:
@liamfraser and @aallan - Yes, this was the exact issue! THANK YOU!

I've been banging my head for a few days on this and that completely solved the issue.

I've now implemented double-buffering support also in the code at: https://github.com/shreeve/pico-usb/blob/main/host/host.c -- I still need to review and test it, but it's essentially there. I just need to beat it up and make sure it's solid.

On a related question, when double-buffering *isn't* needed then the RP2040-E4 errata issue arises where:

> The USB host controller has a bug (RP2040-E4) that means the status written back to the buffer control register can appear in the wrong half of the register. Bits 0-15 are for buffer 0, and bits 16-31 are for buffer 1. The host controller has a buffer selector that is flipped after each transfer is complete. This buffer selector is incorrectly used when writing status information back to the buffer control register even in single buffered mode. The buffer selector is not used when reading the buffer control register. The implication of this is that host software needs to keep track of the buffer selector and shift the buffer control register to the right by 16 bits if the buffer selector is 1.

Another description in the datasheet says:

> The USB host maintains a buffer selector which switches between BUF0 and BUF1. This should only be toggled in double buffered mode but is toggled in single buffered mode too. For a transaction lasting multiple packets (i.e. length more than 8 bytes in low speed mode, and length more than 64 bytes in full speed mode), the buffer status can be written back to the BUF1 half of the status register when the buffer select is incorrectly set to BUF1. Note this does not affect reading new buffer information from the buffer control register, as the controller ignores the buffer selector in single buffered mode when reading the buffer control register.

And, the workaround is stated as:

> Shift endpoint control register to the right by 16 bits if the buffer selector is BUF1. You can use BUFF_CPU_SHOULD_HANDLE find the value of the buffer selector when the buffer was marked as done.

Even with the above, which I'm sure was not easy to discover nor document, I'm still not 100% certain on the problem or the workaround. Is this basically saying:

> The USB controller can incorrectly write to the **upper half** (bits 16-31) of the *buffer control register*, even in single-buffered mode. To work around this issue, suppose bit 3 of BUFF_STATUS has been set, indicating that the buffer for the EP1_OUT endpoint needs attention. When reading the buffer control register for this endpoint, you would shift its value 16 bits to the right if bit 3 of BUFF_CPU_SHOULD_HANDLE is set.

Is this description correct?

aallan, 2024-02-23:
@liamfraser Are there any documentation changes that need doing as a result of this one?

shreeve, 2024-03-02:
FWIW, here's the list that I came up with for when TRANS_COMPLETE is raised (see datasheet, p. 401):

1. A SETUP packet is sent without {RECEIVE,SEND}_DATA_BITS in SCR
2. A packet is transferred with LAST bit set in BCR
3. A short packet (less than maxsize) is transferred

@aallan - If @liamfraser can confirm this is correct, it would be very helpful to update the datasheet.

Thanks!

liamfraser, 2024-03-11:
Yep, I can get this done. It's easier to update in bulk so let's document the outcome of your #387 issue also.

aallan, 2024-03-11:
Thanks @liamfraser!