raspberrypi/pico-feedback #382: [TYPO] Datasheet: USB Host description

shreeve, 2024-02-23:
@liamfraser @aallan 

[image]

lurch, 2024-02-24:
> I started a markdown file with some of my notes related to USB and creating a USB Host library for the RP2040. I just started today, but I've gleaned from a bunch of sources (I'll add references to them) and have also created a bunch of content and images myself. The link is:
> 
> https://github.com/shreeve/pico-usb/blob/main/rp2040-usb.md

Neat! The videos by Ben Eater are also a fantastic resource:
https://www.youtube.com/watch?v=wdgULBpRoXk
https://www.youtube.com/watch?v=N0O5Uwc3C0o
https://www.youtube.com/watch?v=2lPzTU-3ONI

shreeve, 2024-02-28:
I've got pico-usb USB Host support running now to be able to enumerate several devices, including a Pico Debug probe, an old mouse, a medical device, a flash drive, etc.

This now supports pico-sdk native USB Host mode along with full double-buffering support. Now, it's only enough to do enumeration, but as far as I know, there is NO OTHER CODE that I can find that supports USB Host mode on the rp2040 other than TinyUSB.

Here's the link: https://github.com/shreeve/pico-usb

closed 2026-11-06 by katshann-raspberry

katshann-raspberry, 2026-11-06:
Issue in the pic in the original message will be updated in the next release of the RP2040 and RP2350 datasheets