raspberrypi/pico-feedback #385: USB device erratum for aborted control transfers?

tlyu, 2024-02-25:
It seems that macOS sometimes rapidly issues multiple CDC control requests back-to-back, aborting some of them, when a process closes a CDC serial device. This causes a hard fault in TinyUSB due to buffer control state being retained when a new SETUP packet arrives before the previous control transfer completed.

The datasheet says that the STALL bits get reset by an incoming SETUP packet, but not what happens to the EP0 buffer controls. The USB peripherals on some other MCUs (e.g., STM32) do reset the endpoint state for EP0 when a SETUP packet arrives, regardless of whether the previous control transfer completed.

I don't have detailed traces of what's happening at the hardware level, because I don't currently have a RP2040 board with accessible SWD traces.

I've submitted a workaround to TinyUSB, but this seems like a hardware erratum, because there's no guarantee that the buffer control won't cause an unwanted transaction to occur before the ISR can service the interrupt for the new SETUP and adjust the buffer controls.

See also:

adafruit/circuitpython#8824 (analysis and traces)
hathach/tinyusb#2492 (workaround patch)

aallan, 2024-02-25:
Sounds like one for @liamfraser ?

tlyu, 2024-03-05:
Some tracing added by @eightycc seems to confirm that the USB peripheral doesn't reset anything in the EP0 buffer controls when receiving a SETUP packet.

danielstuart14, 2024-04-26:
I think this is something that I'm seeing on my board, even with the latest tinyusb (after patch). (I also tested with Embassy-USB, same thing).
We have a board that has a USB2422 HUB that connects both an RP2040 (B2 Revision) and an external USB port.

Sometimes, when connecting a new device to the external port, the RP2040 will stop communicating with the O.S. (we are using CDC, device is still recognized, just won't accept any messages - ERR 13 on windows):

[image]

USB2422 Errata says that there is an issue that can happen with split transactions, we checked the RP2040 USB signals and it seems like one packet sometimes looses some of its data:
[image]

The problem is that the RP2040 isn't able to recover from this issue, and just stops acknowledging the setup packets from the HOST:
[image]

If we try to send any new data through the CDC interface, this is what happens in the bus:
[image]

1 - The host sends the data, but no ACK/NACK is received
[image]

2 - The host tries to receive some data (44us later):
[image]

3 - Device responds an ACK (9us later):
[image]

4 - Nothing is sent for 30us and then the HOST retries...

The only things that we found will make it work again are:
- Reseting the RP2040
- Reseting the HUB (by its reset pin)
- Disabling the RP2040 Composite Device in device manager and re-enabling it again.

For the two latter cases, we found out that the USB line reset is what makes it work again.
[image]

@tlyu is this something that your patch should fix? By reading both the description here and in your commit I thought it was similar.

zhangXianMr, 2026-10-13:
## Independent third-party confirmation: RP2350 (ARM-S) SIE lockup matching this erratum, master DCD does not fix it

We (EMV card-simulator firmware, RP2354A, TinyUSB 0.17.0 vendored in pico-sdk 2.1.0) have been chasing a USB device lockup whose trigger pattern matches this issue exactly: host fires rapid consecutive control requests on CDC and abandons some mid-transfer → EP0 buffer control left dirty → SIE stops processing (SOF register freezes), device requires USB reset/replug without in-firmware recovery.

### Two-board controlled experiment (same firmware, same host, same day)

| Board (USB serial) | Behavior under identical diag(32-page) stress loop |
|---|---|
| 52CAF624B468F761 | 8 stress rounds → 5 write timeouts; board-side sofwd/txstall counters increment **in sync with each timeout**; SOF frozen ≥3 s = hardware-level SIE stall evidence. Self-healing chain (tx-stall watchdog → stack reinit → soft reboot) caught every hang, zero manual replug. |
| 6AE7FD16713B4282 (known-good) | 30 min identical stress: **zero lockups**, txstall/sofwd stay 0, uptime monotonic. |

So the failure is board/state-dependent, not purely software-configuration-dependent.

### DCD software layer ruled out

We replaced the entire rp2 DCD triplet (dcd_rp2040.c / rp2040_usb.c / rp2040_usb.h) with **TinyUSB upstream master** (Oct 2026, incl. rp2 common refactor dfcd2714, double-buffer out-of-order fix dd08939f, ping-pong rewrite). Symbol-level verification confirmed the replacement actually linked. Result: **failure frequency identical to vendored 0.17** (same load, txstall=2/sofwd=2 per session). Upstream EP0 fixes do not cure this on RP2350 — consistent with this being a hardware erratum rather than a stack bug.

### In-firmware mitigation that works

A TX-stall watchdog in the application (single-byte probe write every 1 s on CDC1; 3 consecutive stalls → USB stack soft reinit; 6 → chip reset) converts every observed hang into a seconds-level self-heal. We now run this in production OTA.

### Artifacts

- Reproduction shape: host opens/closes CDC port repeatedly while device streams large IN frames; hang occurs in abandoned-control-transfer window (matches @tlyu / @eightycc tracing conclusion that SETUP does not reset EP0 buffer controls).
- We can provide logic-analyzer captures of D+/D- at the freeze moment (planned next step) and our full incident writeup + before/after counter telemetry if useful to maintainers.

Question for @liamfraser: is there any known hardware workaround short of USB peripheral reset — e.g. does a double-buffer ABORT on EP0 IN/OUT followed by re-arm recover the SIE deterministically in silicon-rev-1 RP2350? Our soft-reinit works but we'd prefer a narrower recovery path.