raspberrypi/pico-feedback #387: [DOCUMENTATION] No information on how to use RESET_BUS in USB Host mode

shreeve, 2024-02-29:
`Table 403. SIE_CTRL Register` near page 400 (Section 4.1) shows that `Bit 13 - RESET_BUS` will reset the bus. However, there is no documentation on how this works or what to do and TinyUSB doesn't do anything (the line of code says "// TODO: Nothing to do here yet. Perhaps need to reset some state?").

Can documentation be added to show how to use RESET_BUS in USB Host mode?

shreeve, 2024-03-02:
Thanks @aallan -- Just to check on this, is this an ok place to ask the question? I don't see any reference in the datasheet and I'm not sure where else to ask. The TinyUSB code has basically a nop for the USB bus reset. I'm not sure who else to ask, since I don't see it in the documentation and the reference implementation doesn't use or mention how to do a USB bus reset.

shreeve, 2024-03-06:
[image]

aallan, 2024-03-07:
@liamfraser is probably the person to ask.

lurch, 2024-03-07:
(...who is currently on holiday this week :wink: )

shreeve, 2024-03-11:
Excellent information! Thank you.

shreeve, 2024-03-11:
Is there any use for the `RESET_BUS` then?