raspberrypi/pico-feedback #388: Error in wget command in "Getting started with Raspberry Pi Pico"

davidamis, 2024-03-06:
On page 4 of "Getting started with Raspberry Pi Pico" there is a command to get the current copy of pico_setup.h:

wget https://raw.githubusercontent.com/raspberrypi/pico-setup/master/pico_setup.sh

It works correctly the first time, you will get:
    pico_setup.sh
However, if you run it again to reinstall, you will get:
    pico_setup.sh      (version when first downloaded)
    pico_setup.sh.1    (current version)

From the wget man page:
When running Wget without -N, -nc, -r, or -p, downloading the
           same file in the same directory will result in the original
           copy of file being preserved and the second copy being named
           file.1.

The instructions in "Getting started with Raspberry Pi Pico" and/or the pico_setup.sh seem fragile, I had to re-install to get the SDK to work properly, which is how I discovered this. It's important to fix this, as any changes to pico_setup.sh on github won't be used if the manual's instructions had been previously followed.

I'd also suggest a "clean install" section be added to the document.

A further test for the developer of the script would be to compare the filesystem of a first install to a second install to see if there are any other cases like this.

lurch, 2024-03-07:
Note that `pico_setup.sh` itself may not work correctly if you run it more than once - see https://github.com/raspberrypi/pico-setup/issues/10

aallan, 2024-03-07:
Yup. The script was never really designed to be run twice. I just re-read the appropriate place in the [getting started guide](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf). There's no suggestion there that it can be re-run.

I'll add some additional text emphasising that it should be run from a clean installation of Raspberry Pi OS.

closed 2024-03-07 by aallan

davidamis, 2024-03-07:
So, you're not going to fix the wget command in the document? It only requires adding a flag to the command.

As I said, the reinstall did work to fix the problem. The only detail I left out is that I deleted the "pico" folder before reinstall.

lurch, 2024-03-07:
`pico_setup.sh` get updated fairly infrequently https://github.com/raspberrypi/pico-setup/commits/master/ so I don't think this is a massive concern.