raspberrypi/pico-feedback #389: Documentation needs update on DDEBUG_ON_PICO=ON

bfloch, 2024-03-28:
I followed v2.3 of the Appendix at:
https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf

This resulted in
```
Error: Failed to connect multidrop rp2040.dap
```

The fix is to configure the "picoprobe" (debugprobe) build with `-DDEBUG_ON_PICO=ON` as the readme here suggests.

This change should be updated in the documentation.
Sorry if this is the wrong tracker for the issue. I was unable to find the source of the document, so hopefully it will get noticed here.

Thanks.

aallan, 2024-03-28:
I think this has already been fixed internally, and will be in the next documentation release? @nathan-contino I think that was you, is that right?

nathan-contino, 2024-04-29:
Fixed internally, rolling out a new update soon!

closed 2024-04-29 by nathan-contino