raspberrypi/pico-feedback #397: Latest datasheet missing changelog

thejpster, 2024-08-18:
I'm looking at datasheet:

```text
build-date: 2024-08-16
build-version: 4c00622-clean
```

It's missing an update to *Appendix H: Documentation Release History* so I can't see what changed since 2024-08-08.

nathan-contino, 2024-08-19:
Thanks for bringing this up. While we've released a couple of revised versions of the datasheet since release day, the changes have been mostly relegated to typos and minor one-line consistency fixes that weren't worth a changelog entry. When we revise and expand anything significant in the future, I'll be sure to add a changelog so you can follow along with our changes.

Apologies that we haven't been too transparent with the minor follow-up fixes. There's only one of me, so during a major release like this, I was forced to prioritise.

closed 2024-08-19 by nathan-contino

thejpster, 2024-08-20:
Having laboriously diffed the two documents, I observe the following non-trivial modifications:

1. *Chapter 1*: removal of the line:
  > Extended low-power sleep states with optional SRAM retention: as low as 10 μA DVDD
2. *Chapter 1*: removal of the line:
  > GPIOs are 5 V-tolerant (powered), and 3.3 V-failsafe (unpowered)
3. *Chapter 14, Section 14.8.2.1 Pin Types*: removal of the line:
  > Additionally, they will tolerate voltages up to 5.5V, provided IOVDD is powered to 3.3V.

Seems a bit more than a typo, and I'm disappointed to see a headline feature quietly withdrawn.