raspberrypi/pico-feedback #399: The Getting Started guide does not document how to -DPICO_BOARD using VS Code

rppicomidi, 2024-08-23:
I am referencing [The Getting Started Guide](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf) as of  23-Aug-2024. The document is clear on how to use the command line to change `PICO_BOARD`, but I found no instruction on how to change it using VS Code.

I realize the [Raspberry Pi Pico VS Code Extension](https://github.com/raspberrypi/pico-vscode) is changing fast, but at some point, the document should be revised to catch up.

As of extension version 0.15.2, there is a `Change Board` option under the extension's `Project` heading. For 0.15.1 or earlier, you can do this:
1. On the `PROJECT STATUS` line of the CMAKE window, click the `Open CMake Tool Extenstions Settings` gear icon. You have to mouse over  the `PROJECT STATUS` line for the icon to appear.
2. In the new Settings tab that opened in the editor pane, click the `Workspace` tab.
3. Scroll down to the `CMake:Configure Args` item and click the `Add Item` button.
4. Enter `-DPICO_BOARD=[your board name goes here]` and click OK. The board name is the file name without
       extension from `${PICO_SDK_PATH}/src/boards/include/boards`. For example, if you are using a
       adafruit_feather_rp2040_usb_host board, you would enter `-DPICO_BOARD=adafruit_feather_rp2040_usb_host`

Either workflow makes the appropriate changes to the `.vscode/settings.json` file.

lurch, 2024-08-23:
> I realize the [Raspberry Pi Pico VS Code Extension](https://github.com/raspberrypi/pico-vscode) is changing fast, but at some point, the document should be revised to catch up.

Yes, I'm sure that'll happen at some point when things have stabilised a bit.

> As of extension version 0.15.2, .... For 0.15.1 or earlier, ....

Huh, doesn't seem to be much point in documenting old versions, if the feature has already been replaced / improved in the current version? :thinking:

rppicomidi, 2024-08-23:
I suppose not. Just mentioning it because I have one PC that is not using the new VS Code extension (set up using the old
Getting Started guide) but still needs to set the CMake configuration in the `settings.json` file.