raspberrypi/pico-feedback #400: My first Pico 2 SDK 2.0 issue

Some0neS0mewhere, 2024-08-24:
Hi All,

I've just run through the SDK Getting started guide using a Pico 2 and SDK 2.0 under visual studio code as mentioned in the getting started guide. When running the debug routine using a pico debug probe the following is reported for the blink project:

[image]

will-v-pi, 2024-08-24:
Yes, this is a known issue with the Arm toolchain gdb on Windows - but it shouldn’t actually affect functionality. Other than this warning, did the debugging work fine?

Some0neS0mewhere, 2024-08-25:
Hi will-v-pi,

Yes all seems to work fine with the Pico 2 under vs code and the Pico extension (SDK 2) when trying the blink example. 

I am having another issue however with blink and blink_w when switching to either the original Pico and/or the Pico W. Perhaps this issue is meant for a different platform.

Cheers

lurch, 2024-08-25:
Yeah, issues with picotool should be reported at https://github.com/raspberrypi/picotool/issues please.