raspberrypi/pico-feedback #403: why exception priority of SVCall/PendSv/SysTick use different register?

oldoldman, 2024-08-28:
Hi guys,
From the ARMv6-M doc, the algorithm used to calculate the exception priority is as follow (copied from ARMv6-M page B1-195)
```
integer ExceptionPriority[integer n]
assert n >= 1 && n <= 48;
if n == Reset then result = -3;
elsif n == NMI then result = -2;
elsif n == HardFault then result = -1;
elsif n == SVCall then result = UInt(SHPR2.PRI_11);    // A
elsif n == PendSV then result = UInt(SHPR3.PRI_14);  // B
elsif n == SysTick then result = UInt(SHPR3.PRI_15);   // C
elsif n >= 16 then
r = (n - 16) DIV 4;
v = n MOD 4;
result = UInt(NVIC_IPR[r]<v*8+7:v*8+6>); // External interrupt (n-16)
else
result = 4; // Reserved exceptions
assert -3 <= result && result <= 4;
return result;
```
it seems priority of SVCall/PendSv/SysTick is read from SHPR2/SHPR3 register, why not just use NVIC_IPR?

lurch, 2024-08-28:
I dunno the answer to your question, but I do see that SHPR2/SHPR3 are documented in https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf

[image]

> From the ARMv6-M doc, the algorithm used to calculate the exception priority is as follow (copied from ARMv6-M page B1-195)

Is that maybe a question for the Arm engineers, rather than the Raspberry Pi engineers? :shrug:

kilograham, 2024-09-03:
this is an arm question

closed 2024-09-03 by kilograham