raspberrypi/pico-feedback #414: RP2350 clock speed

simonjwright, 2024-09-19:
I wanted my Pico 2’s default architecture to be risc-v, so set CRIT1_BOOT_ARCH (`picotool opt set crit1.boot_arch 1`).

The chip now runs at one third the nominal rate (50 MHz instead of 150).

simonjwright, 2024-09-28:
I now think that the chip was running of the USB clock (48 MHz).

I should have added that I’m trying to create a bare-metal Ada runtime, so I need to understand how to start the chip up properly; I’m about to try to work out how the SDK does it.

Before I made the CRIT1_BOOT_ARCH  change, I was switching to risc-v by running a program that someone posted called `no_flash.uf2` which presumably ran the clock up properly; I’d load my program under GDB, so no clock resetting. Without that, the chip managed to run off the USB clock.

Chalandi, 2024-09-29:
Hi @simonjwright, I created also a bare-metal C/C++ runtime for pico2 (no SDK). 

I faced similar issue during the setup of the PLL to 150MHz on RISC-V, the issue was that the value of the divider CLK_SYS_DIV.INT is somehow read as zero if I modify it after enabling the PLL.

The datasheet saying that field could be changed on-the-fly but apparently is not the case on RISC-V, and I did not get that issue on CM33.

so the workaround was to set the value of the divider CLK_SYS_DIV.INT before enabling the PLL on RISC-V.

I configured the PLL to 150 MHz, you can find the code here:

https://github.com/Chalandi/Blinky_Pico2_dual_core_nosdk/blob/master/Code/Mcal/Clock/Clock.c

Hope this could help you.

simonjwright, 2024-09-29:
@Chalandi, thanks very much for this!

I reworked my order of operations to match yours (I didn't actually check your suggested change, I just did it).

My code still crashed the JTG connection.

It turned out I had (effectively)
```
HW_PER_CLOCKS->CLK_SYS_CTRL.bit.SRC    = CLOCKS_CLK_SYS_CTRL_SRC_clksrc_clk_ref;
```
(the default in the SVD); your
```
HW_PER_CLOCKS->CLK_SYS_CTRL.bit.SRC    = CLOCKS_CLK_SYS_CTRL_SRC_clksrc_clk_sys_aux;
```
worked fine.