raspberrypi/pico-feedback #418: Fritzing file for Raspberry Pico WH?

mgh128, 2024-09-26:
Many thanks for providing Fritzing Part file for the Raspberry Pico W at 
https://datasheets.raspberrypi.com/picow/PicoW-Fritzing.fzpz

Could someone please also provide the corresponding Fritzing file for the Raspberry Pico WH ?

nathan-contino, 2024-10-04:
@JamesH65 if there's any chance you can whip up a Fritzing part that captures those subtle differences, I'm happy to add it to the documentation. If we'd prefer not to create a separate part for such an insignificant difference, let me know here so I have it on paper in case this request comes up again 😄

JamesH65, 2024-10-04:
For the Pico 2 we deliberately made a generic part for Pico and Pico 2 to reduce the update burden and make the docs that support both easier to deal with (i.e one diagram needed). But it is annoying the H variants have SWD in a different place.