raspberrypi/pico-feedback #422: Pico not showing as comport
Even after uploading the  "hello_serial"

jk-aero, 2024-10-14:
I tried uplading most of the example files which have code to print something through serial port nothing works.

Also i tried the blink.uf2 ans its working fine
The code is being execute but port is not showing

Pico works fine.

i tried uplading the micropython.uf2 and then its showing up on comport.

I also triex uplading the flash nuke.uf and tried the example files it didnt work.

lurch, 2024-10-27:
Sorry to sound a bit glib, but many other people have got this working just fine. I can only suggest you double-check https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf and then ask for help at https://forums.raspberrypi.com/viewforum.php?f=144 if you're still having problems.

Please **do** let us know if you think that there any parts of the documentation that could be improved though.

peterharperuk, 2025-07-09:
Sorry, must be an issue at your end with your PC.

closed 2025-07-09 by peterharperuk