raspberrypi/pico-feedback #423: RP2350 Atomic Register Access

adfernandes, 2024-10-17:
In the `2024-10-16` build of the RP2350 Datasheet, under [Section 2.1.3 Atomic Register Access](https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf#2.1.3.%20Atomic%20Register%20Access) there is a description of which peripherals support Atomic Register Access.

Specifically, that section states:

> The following registers do not support atomic register access:
• SIO (Section 3.1), though some individual registers (e.g. GPIO) have set, clear, and XOR aliases
• Any register accessed through the self-hosted CoreSight window, including Arm Mem-APs and the RISC-V Debug
Module
• Standard Arm control registers on the Cortex-M33 private peripheral bus (PPB), except for Raspberry Pi-specific
registers on the EPPB
• OTP programming registers accessed through the SBPI bridge

This description is, to me, somewhat imprecise and I am having a hard time understanding it.

**If my understanding is correct**, the **only** peripherals/registers that **support** atomic register access _via MCU core_ are:

* all read/write registers in the `0x40000000 APB Peripherals` block
* all read/write registers in the `0x50000000 AHB Peripherals` block
* ARM access to the read/write registers in the core-private `0xe0080000 EPPB` block

_(I am targeting multi-core applications, **not** debug access!)_

Is my understanding correct?

Thank you for producing such an amazing document, by the way, and _serious_ hat-tip to _everyone_ who worked on that document... it's one of the best MCU datasheets I've _ever_ read!

nathan-contino, 2024-10-18:
@Wren6991 is the expert on this, I believe. Let's see what he has to say.

I'll keep my 👀 on this conversation because it sounds like we might be able to adjust the language in this section to clarify some ambiguity.

Thanks for the kind words! Our engineers put a lot of effort into this datasheet, all to help users like you wrap your heads around our hardware without melting your brains and/or bashing your heads against the wall. Barring some ambiguity, we _mostly_ accomplish that :-)

Wren6991, 2024-10-18:
> If my understanding is correct, the only peripherals/registers that support atomic register access _via MCU core_ are:

>    all read/write registers in the 0x40000000 APB Peripherals block
>    all read/write registers in the 0x50000000 AHB Peripherals block
>    ARM access to the read/write registers in the core-private 0xe0080000 EPPB block

This is almost correct, except that the self-hosted access to the APs is in the `0x4` APB space, but does not support atomic modifies.

[image]

I'm also confused by the emphasis on "via MCU core" -- all that matters is the address (and its accessibility). It doesn't matter whether the access originates from a processor, the debugger, or the DMA.

adfernandes, 2024-10-18:
Thank you, @nathan-contino and @Wren6991!

@Wren6991 you _completely_ answered my question and even managed to figure out my somewhat mangled wording.

> I'm also confused by the emphasis on "via MCU core"

Ah yes, the way that you put it - that atomic access is a property of the _peripheral_, not the accessor - makes _way_ more sense than what **I** was trying to say!

I was trying to express the fact that the EPPB block will generate a bus fault if accessed from the RISC-V cores, so "don't do that".

My awkward wording comes from the fact that I'm extending the `rp235x-pac` crate to add an `AtomicallyModifiable` trait, so I was trying to say "don't add that trait to the EPPB block" for the `risc-v` functions... but I just realized that the API probably doesn't generate accessors for that block on `risc-v`!

_(I just checked, and [**oopsie** yes it does](https://github.com/rp-rs/rp235x-pac/blob/ecd2c067f9080e1d948192dff2961c6e9be8f1c0/src/inner/mod_risc_v.rs#L754) ... time to [file a bug report!](https://github.com/rp-rs/rp235x-pac/issues/6)_

-----

Anyhoo, I can bit-bang registers and peripherals all day long, but I've grown fat and lazy letting Segger or `pyOCD` set up `CoreSight` for me, so all that stuff is hazier for me than it should be, and I forgot all about the **self-hosted debug** stuff, mainly because I haven't seen that on any of the other Cortex-M chips I've worked with.

But of course, the section in the datasheet is a _fantastic_ resource, so again - thank you!

-----

Lastly, here is my pseudo-code for the above discussion, **please give it a look-see** and feel free to add it to the next datasheet spin, if you think it's correct/useful!

```c
bool has_atomic_register_access(const uint32_t register_address) {

    /// assume that `register_address` is a valid, 32-bit aligned peripheral register address

    if (is_not_read_write(register_address)) return false; // the register must have `read-write` access

    if (is_ahb_peripheral(register_address)) return true; // is in the `0x50000000 AHB Peripherals` address block

    if (is_apb_peripheral(register_address)) { // is in the `0x40000000 APB Peripherals` address block block

        if (is_self_hosted_debug(register_address)) {
            // exclude all `CORESIGHT_*` APB registers noted in section "2.2.4. APB Registers"
            // of the datasheet, which are addresses `0x40140000` to `0x4014afff`, inclusive
            return false;
        }

    }

    if (is_extended_private_peripheral(register_address)) {
        // exclude all EPPB peripherals noted in section "3.7.5.1. Cortex-M33 EPPB Registers"
        // of the datasheet, which are addresses `0xe0080000` to ` 0xe008000b`, inclusive
        return false;
    }

    return false;

}
```