raspberrypi/pico-feedback #431: Bus Priority on the RP2350 vs RP2040

XZCE, 2024-12-02:
In my Pico W project (RP2040) I use the following code to make sure the 2nd CPU doesn't get stalled (it's serving the hardware bus of an old retro computer):

bus_ctrl_hw->priority = BUSCTRL_BUS_PRIORITY_PROC1_BITS;

I've obtained some Pico 2Ws (RP2350), compiled my project for them, and I've tracked down CPU0 being so sluggish that WIFI isn't working reliably (LED isn't reliable, connections aren't either) and SD Card reads are very slow.. to this statement.

My question is, is this no longer an appropriate thing to use on the RP2350 when expecting WIFI to work?  My thoughts are that the M0+ didn't have enough grunt to soak up resources to a point of failure, but maybe the M33's do?

Thanks for any advice.

XZCE, 2024-12-03:
Remove the priority setting to get the Pico 2W's LED blinking again...

```c
#include "pico/stdlib.h"
#include "pico/cyw43_arch.h"
#include "hardware/gpio.h"
#include "hardware/structs/bus_ctrl.h"

void setup () {
    pinMode (LED_BUILTIN, OUTPUT);
}

void loop () {
    delay (200);
    digitalWrite (LED_BUILTIN, HIGH);
    delay (200);
    digitalWrite (LED_BUILTIN, LOW);
}

void setup1 () {
    // Put this CPU(1) into high priority bus mode
    bus_ctrl_hw->priority = BUSCTRL_BUS_PRIORITY_PROC1_BITS;
}

volatile uint32_t data [8];

void loop1 () {
    while (1) {
        data [0] = sio_hw->gpio_in;
        data [1] = sio_hw->gpio_in;
        data [2] = sio_hw->gpio_in;
        data [3] = sio_hw->gpio_in;
        data [4] = sio_hw->gpio_in;
        data [5] = sio_hw->gpio_in;
        data [6] = sio_hw->gpio_in;
        data [7] = sio_hw->gpio_in;
    }
}
```

lurch, 2024-12-03:
Based on the `setup()` and `loop()` function names above I'm guessing that you're using one of the Arduino wrappers?
Please recreate this using "just" `pico-sdk` as that'll make it much easier for us to investigate.

XZCE, 2024-12-04:
Yes mate, I am.  Here's the SDK version, archive attached.
[blink.zip](https://github.com/user-attachments/files/18002809/blink.zip)