raspberrypi/pico-feedback #432: Documentation missing 'sudo make install' for OpenOCD

themzlab, 2024-12-02:
in this document, as of the 15 October 2024 update...
[Getting Started With Pico](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf)

Appendix A
building OpenOCD

installation instructions are missing a required line:

**sudo make install**

and optional flags need to be specified during the configuration step. I found these in the setup_pico.sh script
--enable-ftdi --enable-sysfsgpio --enable-bcm2835gpio --disable-werror

however, when I built it I ended up using these options and it has worked so far for my configuration to load .elf files with the Raspberry Pi Debug Probe
--disable-werror --enable-cmsis-dap

before I found this repository, I made post in this forum about it.
[pico sdk forum post](https://forums.raspberrypi.com/viewtopic.php?t=380311)

lurch, 2024-12-02:
I believe the `--enable-sysfsgpio --enable-bcm2835gpio` options are needed if you're wanting to use the GPIO pins on a Raspberry Pi to debug a Pico (which was the recommended way of doing things when Pico 1 was first released). However that was before DebugProbe existed, and therefore the instructions for directly wiring a Raspberry Pi's GPIO pins to a Pico have now been removed from the Getting-Started guide.

I also see that Appendix A says "To start openocd from the build directory, you can use:" which implies that `sudo make install` isn't actually necessary? :thinking:

lurch, 2024-12-02:
Also, I see that your forum thread says "if you install it with apt, then you will not get the build that comes from Raspberry pi".

For Raspberry Pi OS, we **do** ship our custom version of openocd inside the apt-installable `openocd` package. It's only if you're using another OS e.g. Ubuntu that you'd get the "upstream" version of openocd and not our customised version. However we _have_ been trying to push our modifications to the source-code upstream too, so at "some time in the future" even Ubuntu's package of openocd will include our changes (to add support for RP2040 and RP2350).

I hope that all makes sense! :slightly_smiling_face:

themzlab, 2024-12-02:
what I can say is I had a really rough time getting things to work until I did the **sudo make install** command.  and I only realized I needed it after I found the existence of pico_setup.sh and looked inside.  That script is mentioned later in Appendix C.

in Appendix A there is a segment titled "Debug with OpenOCD and the following command is suggested:
sudo openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "adapter speed 5000" -c "program blink.elf verify reset exit

that line is not going to work without specifying full paths to the .elf or full paths to the two .cfg files.  It will not work as is if you have followed the build instructions without the install command.

I don't know how many people are going to realize that build+install or install from apt are two optional ways of doing something that is similar.  People might be familiar with embedded compilers and know little about building packages on a linux system.

LostinTimeandspaceYT, 2024-12-10:
Just wanted to add that I also ran into this issue after coming back to a project I updated in September of this year.   I agree with @themzlab that the documentation should add some additional notes regarding the upstream discrepancy in `openocd`.

I wrote a [bash script](https://github.com/LostinTimeandspaceYT/pico_drivers/blob/trunk/tmux_debug.sh) that uses `tmux` to create a debugging environment, which worked great up until the recent changes. 
```sh
# Create a pane for OpenOCD 
# NOTE: Check the README for information regarding openocd.
# AS of Dec 10, 24: Openocd must be built from source.
tmux send-keys -t pico_debug "sudo openocd -s tcl -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c \"adapter speed 5000\"" Enter
```
we didn't use to have pass in `-s tcl`, and the documentation does not explain the need for the change from previous versions. As @themzlab stated, without running `sudo make install`, full paths are needed to get openocd to work properly. 

I tried adjusting my `sudoers` to allow the custom version of openocd to run with sudo without a password, but that no longer works. 

Hopefully the changes can be upstreamed soon!

## System Details:
Pop_OS! (Ubuntu 22.04)
Ryzen 5 3600X
32GB RAM