raspberrypi/pico-feedback #435: Flashing Pico via Rasperry Pi

hasenradball, 2024-12-14:
Dear Docu Team,

why did you remove the docu part how to flash and debug the pico directly via pi?
It is so great doing so!!!

hasenradball, 2024-12-15:
One other point to this topic.

It would be perfect for the docu if there would be an example task in VScode for:
a) flashing the pico
b) starting the debuging

Is there an hint for the old docu version?

lurch, 2024-12-16:
> It would be perfect for the docu if there would be an example task in VScode for: a) flashing the pico b) starting the debuging

Did you forget to read Chapter 4 of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf ?
ping @will-v-pi in case I've misunderstood the request.

> Is there an hint for the old docu version?

We don't provide old versions of the documentation, and we only provide support for what's written in the latest version of the documentation (which can always be found at https://datasheets.raspberrypi.com/ ).

hasenradball, 2024-12-16:
@lurch  @will-v-pi 
What I mean here maybee wrongly explained…

with the extension in vsCode, will it also possible to flash and debug  the pico via raspberry pi directly?
I set up openocd to flash an debug the pico via 40 pin header of pi.

Will this already be possible?

lurch, 2024-12-16:
> with the extension in vsCode, will it also possible to flash and debug the pico via raspberry pi directly?

As I've already tried to suggest, this is no longer directly supported; we only support using Debug Probe.
If you're able to get it working, then great we aren't going to stop you, but I'm afraid we won't be providing help with this.

Closing the issue.

closed 2024-12-16 by lurch

hasenradball, 2024-12-16:
understood.