raspberrypi/pico-feedback #436: [chore] RP2350 datasheet, PIO pseudocode in 11.5.4.1. Autopush Details is wrong

naoppy, 2024-12-23:
RP2350 datasheet: https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf
On p904, Section 11.5.4.1. Autopush Details
sample code says
```python
 1 isr = shift_in(isr, input())
 2 isr count = saturate(isr count + in count)
 3
 4 if rx count >= threshold:
 5   if rx fifo is full:
 6     stall
 7   else:
 8     push(isr)
 9     isr = 0
10     isr count = 0
```
But I think line 4 should be `4 if isr count >= threshold:`.
I'm not an expert of pico, so I might be wrong.

naoppy, 2024-12-23:
And more, I found 1 more typo.
p905, last line, says `"PULL does not require similar behaviour, because autopush does not have the same nondeterminism."`
But it should be `"PUSH does not require similar behaviour, because autopush does not have the same nondeterminism."`

lurch, 2024-12-23:
> RP2350 datasheet: https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf On p904, Section 11.5.4.1. Autopush Details sample code says
>
> ```python
>  1 isr = shift_in(isr, input())
>  2 isr count [...]

I also don't know enough about PIO to know if this is an error or not. But if it is wrong, the same mistake is present in section 3.5.4.1. of the [RP2040 datasheet](https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf).

katshann-raspberry, 2026-11-06:
This is updated in the latest release of the RP2350 datasheet and will be included in a future release of the RP2040 datasheet.

closed 2026-11-13 by katshann-raspberry