raspberrypi/pico-feedback #441: picotool missing from Debian Testing

thejpster, 2025-01-07:
`picotool` has been [dropped from Debian Testing](https://tracker.debian.org/news/1601076/picotool-removed-from-testing/), [breaking my CI pipelines](https://github.com/rp-rs/rp-hal/actions/runs/12659014824/job/35277176294) for Rust on Raspberry Pi.

Is it coming back, or should I try and get picotool from somewhere else?

jannic, 2025-01-07:
It should come back automatically once https://buildd.debian.org/status/package.php?p=gcc-riscv64-unknown-elf (which is an indirect dependency) is built and uploaded for all architectures.

lurch, 2025-01-08:
I'm afraid that Debian's packaging decisions are outside of Raspberry Pi's remit, so I'm going to close this issue.

closed 2025-01-08 by lurch

thejpster, 2025-01-09:
Ok, but do you have a recommendation on where CI pipelines can get pre-compiled copies of picotool from? Or do you strictly supply it as source code only?