raspberrypi/pico-feedback #442: FreeRTOS ports have been upstreamed

lurch, 2025-01-09:
In section 2.3.7 of https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf there's a **NOTE** which says:
"As of the time of press the latest FreeRTOS-Kernel change for SDK 2.0.0 and RP2350 have not been merged upstream; they can be found [here](https://github.com/raspberrypi/FreeRTOS-Kernel) instead."

However this is no longer true, the FreeRTOS ports _have_ now been upstreamed - see https://github.com/raspberrypi/pico-sdk/issues/1453#issuecomment-2580099112

jondurrant, 2025-02-11:
Thanks Andrew.

RP2350 support is not included in the current head of FreeRTOS Kernel or the current release which V11.1.0 and predates RP2350 release. So I am assuming this Repo is still current and supported. No mention of it being deprecated in the README that I spotted.

I'm confused now....

lurch, 2025-02-11:
Copy'n'pasting a comment from the issue that I linked to earlier (which in hindsight I should have done earlier)...

> For the benefit of anyone else reading this issue: the FreeRTOS ports have now been upstreamed to https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/main/portable/ThirdParty/GCC/RP2040 and https://github.com/FreeRTOS/FreeRTOS-Kernel-Community-Supported-Ports/tree/main/GCC/RP2350_ARM_NTZ and https://github.com/FreeRTOS/FreeRTOS-Kernel-Community-Supported-Ports/tree/main/GCC/RP2350_RISC-V

jondurrant, 2025-02-11:
Thank you. I suggest the README is updated here with this info.

kilograham, 2025-02-11:
You need to initialize submodules in FreeRTOS (upstream) for RP2350 support, as FreeRTOS no longer wants community/partner ports in the main repo.

The newest `FreeRTOS_Kernel_import.cmake` which you can find in `pico-examples` or in fact now (as of yesterday https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/1243) in both FreeRTOS-Kernel RP2040 or in the submodule RP2350 ports, looks in both possible locations (and thus works for both deprecated RPi repo, and the upstream repo)

We have a little new documentation going into next version of the SDK book, and will also PR some README improvements to upstream FreeRTOS-Kernel

jondurrant, 2025-02-16:
Just to be clear right now if you use https://github.com/FreeRTOS/FreeRTOS-Kernel/ and the latest release version (not head of repo which would in my view be bad practice) which is V11.1.0 the Community-Supported-Ports/GCC is a previous version to that including RP2350 support. So my perception reading this is that RP2350 is dropped as a viable FreeRTOS target until the next release.

Of course, we could potentially use an out-of-line Community-Supported-Ports repo version in a seperate folder outside of the FreeRTOS-Kernel hierarchy. This needs a rewrite of some of the make files though as I see no way to set FREERTOS_KERNEL_RP2040_RELATIVE_PATH manually. Unless I am missing something.

kilograham, 2025-02-16:
ah, i didn't realize that - i have always used the tip of FreeRTOS!

BryanCrotazGivEnergy, 2025-06-25:
What's the current status? I'm really confused as to which repo I should be using, as I can't get either of them to compile in my project

kilograham, 2025-06-25:
You should use the upstream version, but you must initialize the sub-modules if using RP2350 not RP2350 as FreeRTOS requires all new ports to be there.

Also if your project is using `FreeRTOS_Kernel_import.cmake`, make sure you are using a recent version (e.g. copy https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/portable/ThirdParty/GCC/RP2040/FreeRTOS_Kernel_import.cmake)

kilograham, 2025-06-25:
```
git clone git@github.com:FreeRTOS/FreeRTOS-Kernel.git
cd FreeRTOS-Kernel
git submodule update --init
```

BryanCrotazGivEnergy, 2025-06-25:
Ah thanks. There is so much out of date information on this! It compiles now... so ship it?!

akohlsmith, 2026-01-15:
> ```
> git clone git@github.com:FreeRTOS/FreeRTOS-Kernel.git
> cd FreeRTOS-Kernel
> git submodule update --init
> ```

There is a copy of FreeRTOS in pico-sdk under `lib/tinyusb/lib/FreeRTOS-Kernel` -- is it possible to make sure that `git submodule update --init` is executed automatically by cmake?