raspberrypi/pico-feedback #445: Restore Raspberry Pi Pico W Serial Drivers to Factory Settings

ThiagoSousa81, 2025-01-22:
I recently got a BitDogLab board, which has a Raspberry Pi Pico W as a microcontroller. I'm configuring it to run scripts in C. I saw an installation tutorial that teaches how to use Zadig to install the drivers for Interface 0 as USB Serial and Interface 1 as WinUSB. The problem is that when I put the USB Serial driver on interface 0, the serial only appears when the board is in bootsel and is not running the software. When the board is out of bootsel, Windows does not recognize it. I can't find the Serial working in any tested Serial client (PuTTY, VsCode...). I would like to know a way to restore these drivers to the factory settings if possible. I searched a lot on the internet, in the documentation and in forum posts, but I couldn't find anything about it. Some even told me that I shouldn't change the driver for Interface 0, but the tutorial said to do so.

As for other programs, I can easily control the GPIOs and upload .uf2 files, only Serial has this problem. If you can help me I would really appreciate it.

ThiagoSousa81, 2025-01-23:
I followed this tutorial to configure the Pico W. I was unable to download the video lesson as it belongs to the course platform.

[17.37M2 - U3C4O123S - Slides (1).pdf](https://github.com/user-attachments/files/18519484/17.37M2.-.U3C4O123S.-.Slides.1.pdf)

My Pico W is showing up like this in Device Manager when it is in bootsel:

[image]

When it is outside of BootSel it does not even appear.

I put the board in bootsel, compile and run this program here

https://github.com/wedsonjswd/prog02_printf 

But no serial number appears connected

[image]

These are the drivers I am currently using on interfaces 0 and 1

[image]

[image]

I've already tried putting the usbser driver on Interface 0 but the serial port only appears when the board is in bootsel

ThiagoSousa81, 2025-01-23:
This is what happens when I put it in bootsel and open the serial port that appears
[image]

The serial port only appears when I have the usbser driver on interface 0

lurch, 2025-01-23:
I think if you've followed a third-party tutorial, rather than our official documentation, then you'll have to ask that third-party tutorial for help? 😕  (I'm afraid I personally can't help you with Windows driver problems, as I don't use Windows)

Our official documentation can be found at https://www.raspberrypi.com/documentation/microcontrollers/pico-series.html#documentation and some example pre-compiled programs can be found at https://www.raspberrypi.com/documentation/microcontrollers/c_sdk.html#your-first-binaries

lurch, 2025-01-23:
You might be able to get help with Windows-issues by asking for help on the forums? https://forums.raspberrypi.com/viewforum.php?f=143

ThiagoSousa81, 2025-01-24:
Thanks for your help, when I can solve this problem I will post the solution here too

ThiagoSousa81, 2025-01-24:
I managed to solve my problem. What I did was uninstall all the Raspberry Pi Pico W drivers in the Device Manager and let Windows install them automatically.

Then I had to change the interface 1 driver back to WinUSB and the Serial started working when the board was out of the boot.

> This solution only works on Windows 10 or 11

Since the problem has been solved, you can close this issue

[image]

closed 2025-01-24 by ThiagoSousa81